Beaky2000/esphome-p1mini #1: Translate pins to other card

lighty78, 2022-09-23:
I get this error when i connect it and Not changing board:

```
20:49:30 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
-- | -- | -- | --
20:49:36 | [I] | [ota:109] | Boot seems successful, resetting boot loop counter.
20:49:40 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:49:50 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:00 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:10 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:20 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:30 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:40 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:50:50 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
20:51:00 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.

```

I am getting this error when i change board in Yaml-file and tries to install it, and would like help to resolve the correct code/pins for my card.
I use this card: https://i0.wp.com/randomnerdtutorials.com/wp-content/uploads/2019/05/ESP8266-NodeMCU-kit-12-E-pinout-gpio-pin.png?quality=100&strip=all&ssl=1

And i have it connected:

**1 = 5v -> VIN
2 = RTS -> D2 (GPIO04)
3+6= GND/DGND -> left GND
4= Not used
5= TX -> Right GPIO03**

```
INFO Reading configuration /config/esphome/p1mini.yaml...
Failed config

switch.gpio: [source /config/esphome/p1mini.yaml:50]
  platform: gpio
  id: p1_cts
  pin: 
    
    Cannot resolve pin name 'D2' for board esp01_1m.
    number: D2
switch.gpio: [source /config/esphome/p1mini.yaml:54]
  platform: gpio
  id: status_led
  pin: 
    
    Cannot resolve pin name 'D4' for board esp01_1m.
    number: D4
    inverted: True
binary_sensor.gpio: [source /config/esphome/p1mini.yaml:61]
  platform: gpio
  id: secondary_p1_rts
  pin: 
    
    Cannot resolve pin name 'D0' for board esp01_1m.
    number: D0
    mode: INPUT_PULLDOWN
    inverted: False
```

Beaky2000, 2022-09-27:
Try changing "number: D2" to "number: GPIO4", "number: RX" to "number: GPIO3" and "number: TX" to "number: GPIO1". Maybe TX and RX work already and you can keep them as is.

I don't know if your board has an onboard led and which gpio controls it, but change "number: D4" into "number: GPIOn" and if you don't have an led, just use a random free gpio.

D0/secondary_p1_rts is only needed if you want to pass the signal along to a secondary P1 reader, which I assume you do not, so set it to any free gpio. Not all gpio:s support "INPUT_PULLDOWN", but if you try to use one that does not, you will probably get a warning telling you which ones do.

lighty78, 2022-09-27:
Still the original error: 
```

22:07:39 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
-- | -- | -- | --
22:07:39 | [D] | [switch:017] | 'p1_cts' Turning OFF.
22:07:39 | [D] | [switch:037] | 'p1_cts': Sending state OFF
22:07:40 | [D] | [switch:017] | 'status_led' Turning OFF.
22:07:40 | [D] | [switch:037] | 'status_led': Sending state OFF
22:07:40 | [D] | [switch:013] | 'p1_cts' Turning ON.
22:07:40 | [D] | [switch:037] | 'p1_cts': Sending state ON
22:07:40 | [D] | [switch:013] | 'status_led' Turning ON.
22:07:40 | [D] | [switch:037] | 'status_led': Sending state ON
22:07:42 | [D] | [sensor:126] | 'p1mini WiFi Signal': Sending state -73.00000 dBm with 0 decimals of accuracy

22:07:39 | [W] | [p1reader:214] | CRC buffer overrun. Resetting.
-- | -- | -- | --
22:07:39 | [D] | [switch:017] | 'p1_cts' Turning OFF.
22:07:39 | [D] | [switch:037] | 'p1_cts': Sending state OFF
22:07:40 | [D] | [switch:017] | 'status_led' Turning OFF.
22:07:40 | [D] | [switch:037] | 'status_led': Sending state OFF
22:07:40 | [D] | [switch:013] | 'p1_cts' Turning ON.
22:07:40 | [D] | [switch:037] | 'p1_cts': Sending state ON
22:07:40 | [D] | [switch:013] | 'status_led' Turning ON.
22:07:40 | [D] | [switch:037] | 'status_led': Sending state ON
22:07:42 | [D] | [sensor:126] | 'p1mini WiFi Signal': Sending state -73.00000 dBm with 0 decimals of accuracy

```

lighty78, 2022-09-28:
Aidon 6534

Beaky2000, 2022-09-28:
Oh, then you are in luck! Aidon 6534 uses an older binary format instead of the ascii format used by most meters.

However, the binary format was just implemented, so you only have to update to the latest version and everything will work!

lighty78, 2022-09-28:
That is good news! 
I will try in a couple of hours and write again then!

lighty78, 2022-09-28:
YES!  It's working!  Thank you for the binary update! =)

closed 2022-09-29 by Beaky2000