Beaky2000/esphome-p1mini #8: Could not allocate memory for JSON document!

jenshorn, 2022-12-17:
I've had it running for almost a week now but this morning it stopped sending data. The wifi signal is updating and sent to Home assistant but no other values are updated. When viewing the webpage of p1mini state is set to NA on every row and on some occasions i got the error:

`19:48:24 | [E] | [json:043] | Could not allocate memory for JSON document! Overflowed largest free heap block: 8 bytes`

Is this a known issue?

There is pretty bad connection with the Wemos D1 mini in the meter cabinet but the data transfer has been good since the installation. Until today.

Meter: Landis+Gyr E360

Beaky2000, 2022-12-22:
I have not come across this issue before. But some Googling suggests that it is a general issue that can show up in ESPHome in various cases, especially if free memory is limited.

The fact that it ran for a week before this happened suggests that memory is slowly being consumed until not enough remains for normal operation. If you unplug the device and then connect it again, will it run for another week?

jenshorn, 2022-12-23:
Perhaps it might be related to my poor reception somehow? I have improved the Wi-Fi connection (from around -80 to -65) and the device is up and running again. Let's see for how long! Anyway it might be unrelated to the specific implementation of this library as you found it might be ESPHome related instead. Thanks!

vikohub, 2023-04-24:
I also started seeing this after 1day of succesfully running the code on an ESP32-WROOM.
It started suddenly after re-positioning the chip into a different location in my house.
Not sure if an ESP home update in HA coincided.

I also tried with another dev board, and the same result.
Googling suggests a memory leak of an external package - the p1mini.h I suppose in this case.

I tried several fixes suggested at various forums, but nothing helps. Just now I tried compiling with
  framework:
    type: esp-idf
instead of the usual arduino. I'm getting errors related to an int expected, but a long unsigned int received.

"In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/p1mini.h: In member function 'virtual void P1Reader::loop()':
src/p1mini.h:408:38: error: format '%d' expects argument of type 'int', but argument 5 has type 'long unsigned int' [-Werror=format=]
                 ESP_LOGD("p1reader", "Cycle times: Message = %d ms (%d loops), Processing = %d ms (%d loops), (Total = %d ms)",
"

Any thoughts?

Beaky2000, 2023-04-30:
... I guess not.

Anyway, I just made an update which contains some more memory cleanup, but I don't think it will make any difference in practice.

But if anyone sees an error message in the log: "Memory leak detected!", please let me know!

vikohub, 2023-05-02:
Sorry, I was out of office for a while.
Will gladly test now that I am back.

vikohub, 2023-05-02:
At least it starts up fine without the "Could not allocate memory for JSON document" message.
I will test it plugged into the HAN port when I get home.

vikohub, 2023-05-02:
OK, I flashed with the updated p1reader.h, and got a new error - with the same board that I once got working:
"[W] [p1reader: 227]: Unknown data format
(0xDF). Resetting."

This happens when after successfully booting the board with USB powering it, I plug in the rj12 onto the meter's HAN port.
Tried it a few times, and the hex code in the parenthesis changes, but nothing else.
I am wondering is the board broken, or is the meter somehow confused?

Beaky2000, 2023-05-03:
The p1mini uses the first byte it receives to determine if the message is ASCII or binary and there are only two valid values. Any other value will result in the error message you see.

At this point it would certainly be good to test with another p1mini and ideally one that has been verified to work on another meter. Testing your p1mini on another meter would be good too, if possible.

What kind of meter do you have? Is it one of the verified models?

vikohub, 2023-05-03:
Thanks, I will look into that.

While I did manage to get it to work once, perhaps you could reaffirm my thinking as I am running this on a ESP32-WROOM:

Pins are connected as follows:
HAN port:		ESP32:
1 - 5V		----	VIN
2 - rts		----	GPIO14
3 - data gnd 	----	GND
5 - tx		----	RX0
6 - gnd 	        ----	GND

And .yaml:

...
switch:
  - platform: gpio
    id: p1_cts
    pin:
      number: GPIO14
  - platform: gpio
    id: status_led
    pin:
      number: GPIO2
      inverted: true

binary_sensor:
  - platform: gpio
    id: secondary_p1_rts
    pin:
      number: GPIO5
      mode: INPUT_PULLDOWN
      inverted: false

uart:
  id: uart_bus
  tx_pin:
    number: TX
    inverted: true
    mode: OUTPUT_OPEN_DRAIN
  rx_pin:
    number: RX
    inverted: true
    mode: INPUT_PULLUP
  baud_rate: 115200
...

I suppose the pins 1 & 6 are not necessary as I am powering the board from USB, but it did work like this once.

I was not sure on the number: TX/RX on the yaml, and also tried with tx_pin: GPIO1, rx_pin: GPIO3, but homeassistant crashed before I could test that.

Beaky2000, 2023-05-03:
A couple of thoughts:

You shouldn't connect power from the meter and USB at the same time. You probably won't damage anything, but only probably.

Are you connecting the USB to a computer for power? That would explain why you get a lot of "noise" in the communication: The computer UART is connected on the same lead as the meter. The ESP32 does have several uarts, so I think it should work to use GPIO 16 & 17 (while making sure to connect the meter to GPIO16).

vikohub, 2023-05-04:
The meter is Aidon 7534.

I will try powering from USB only, as the HAN port does not provide enough power for an ESP32. I connected the power leads mainly to ensure the board and the port had the same 5V & ground reference.

I did not power the board from computer USB, but a cell phone charger that only delivers power (no data leads).

I will also look into plugging the meter to GPIO16&17, while having the computer USB connected to GPIO1&3 during troubleshooting.

Thanks for your help thus far!

vikohub, 2023-05-04:
Question:
looking at P. Svanstrom's guide on p1reader, it suggests using a pull-up resistor:
"Running on ESP32

The ESP32 requires an external power supply since it cannot be powered by the low amperage available from the P1-port itself. Using the hardware UART on GPIO3 is required.

Use a pull-up resistor (1k) from RXD to 3V3." (I assume that is supposed to read as RXD of the board, which connects to TX of the meter.)

Any reason you do not suggest using one?

I called Aidon, and they suggested connecting a 1 kOhm between TX and 5V, if using a longer cable (5 meters), which I have to use. I tried that, but it did not solve the issue.

I seem to have gotten the board to work (no more JSON errors), and it is reporting wifi signal strength properly - just still no actual meter values.
I will try with a much shorter cable today.

Beaky2000, 2023-05-04:
I do not suggest using an external pull-up because the ESP has an internal pull-up (which is enabled by setting INPUT_PULLUP). This ensures that a good value of resistor that works well with the input is used and also requires no external components.

I see no reason why GPIO3 would be required for the UART. In fact using a separate UART which can not interfere with USB or the logger should be much better! But since the D1mini only has a single UART, I have not tested that.

I do *not* suggest putting a pull-up between TX and 5V, no matter what Aidon claim, because the ESP input is only rated for 3.3V and higher inputs could damage it!

According to this spec, cables lengths up to max 3 meters are supported, so definitely try with a short cable if you are having any kind of problem:
https://www.skekraft.se/wp-content/uploads/2021/03/Aidon_Feature_Description_RJ12_HAN_Interface_EN.pdf

One thing that might be a problem is that the RTS signal from the ESP is only 3.3V and the meter requires minimum 4V according to the spec above. If you are getting absolutely nothing on the UART, this could be the issue and it could probably be solved with a pull-up to 5V and setting the output to OPEN_DRAIN. But since you were getting something, I don't think this is the problem.

vikohub, 2023-05-04:
Hmm, I was also surprised that Aidon advised using 5V for the pull-up, instead of 3,3V.
Although it seems that the risk of damaging the board with 5V input is small:
https://www.qworqs.com/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/

Your theory on the 3,3V RTS being too low is actually quite interesting, since as far as I understand, the wifi-signal measure comes from the board itself, and I seem to get nothing sensible from the meter. Homeassistant shows "Unknown" as the entity states (they go to "Unavailable" if the ESP is disconnected, so some values do come through I guess.)

Does you suggestion translate to this in my setup:

Using 5V from ESP VIN to pull-up RTS; not connecting 5V & GND from meter

[image]

I guess the other alternative would be to connect GDN from meter, and use meter's 5V for pull-up:

Using 5V from meter to pull-up RTS; connecting GND from meter

[image]

Either way, does "setting the output to OPEN_DRAIN." mean this in my .yaml:

```yaml
switch:
  - platform: gpio
    id: p1_cts
    pin:
      number: GPIO14
      **mode: OUTPUT_OPEN_DRAIN**

```
?

Beaky2000, 2023-05-04:
That does look about right. I would use the 5V on the ESP, but it shouldn't really matter.

First I would suggest you set the log level to DEBUG and check in the log if you get anything at all. You can see the log via the web interface in case you didn't know: http://p1mini.local/ (or whatever host name you use)

Are you using a separate uart or the default one? I think the log might show some information about uart configuration on boot.

vikohub, 2023-05-05:
Thanks, I had the DEBUG level on already. It was most helpful.

I am using the default UART.

Surprise surprise, the problem ended up being a bit mystical:
we were doing new electrical cabling runs and I reconnected the board after the cabling was ran. It seems to have interfered with the wifi signal, and perhaps also with the rj12 cable signal. Repositioned the board, added a wifi signal booster that enabled a shorter rj12 cable (now 3 meters), and it works **perfectly** - even without the 1kOhm between VIN & RTS!

I will still test adding that & using OUTPUT_OPEN_DRAIN for the RTS output, and perhaps changing the rx & tx to GPIO16&17 - although now I no longer use the standard tx & rx for USB debugging, so no overlap at the moment.

Thank you so much for your help - I have certainly learned a lot!

After running this setup for a while & confirming that it works stable, I would like to help others struggling with running on ESP32 (and/or needing longer cables) by writing a README subsection draft for ESP32's - if you would kindly agree to add it to your main page after reviewing it?

Beaky2000, 2023-05-05:
I would like to keep the main Readme page clean and focused on the D1mini, but I could see having a separate document with a link from the main Readme with advice for anyone wanting to get it working on alternative boards. Not just ESP32:s, even if that is a good start.

It would be interesting to have instructions for setting up another UART (GPIO 16&17), since that is one (the only?) advantage an ESP32 has over an 8266.

Personally, if the RTS signal is working (and it seems to be since you are getting sensor readings) i would not mess with the external pull-up on the principle of "If it isn't broken, don't fix it".

closed 2023-05-05 by Beaky2000