Beaky2000/esphome-p1mini #11: [W] | [p1reader:227] | Unknown data format (0x00). Resetting.

doe81, 2023-05-28:
#### Issue Description
I get [W] | [p1reader:227] | Unknown data format (0x00). Resetting.
It seemed to happen more often during periods of low load.

* S34U18 (Sanxing SX631) / Vattenfall

#### Log
```
11:51:45 | [D] | [switch:055] | 'p1_cts': Sending state ON
11:51:45 | [D] | [switch:012] | 'status_led' Turning ON.
11:51:45 | [D] | [switch:055] | 'status_led': Sending state ON
11:51:45 | [W] | [p1reader:227] | Unknown data format (0x00). Resetting.
11:51:45 | [D] | [switch:016] | 'p1_cts' Turning OFF.
11:51:45 | [D] | [switch:055] | 'p1_cts': Sending state OFF
11:51:46 | [D] | [switch:016] | 'status_led' Turning OFF.
11:51:46 | [D] | [switch:055] | 'status_led': Sending state OFF
11:51:54 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -72.00000 dBm with 0 decimals of accuracy
11:51:56 | [D] | [switch:012] | 'p1_cts' Turning ON.
11:51:56 | [D] | [switch:055] | 'p1_cts': Sending state ON
11:51:56 | [D] | [switch:012] | 'status_led' Turning ON.
11:51:56 | [D] | [switch:055] | 'status_led': Sending state ON
11:51:59 | [D] | [p1reader:221] | ASCII data format
11:51:59 | [D] | [switch:016] | 'p1_cts' Turning OFF.
11:51:59 | [D] | [switch:055] | 'p1_cts': Sending state OFF
11:51:59 | [D] | [p1reader:282] | CRC verification OK
11:52:00 | [D] | [p1reader:328] | Could not parse value from line '/AUX5U18SEH13F10029'
11:52:00 | [D] | [p1reader:328] | Could not parse value from line '0-0:96.1.0(38303034363937333432383437313035303538393236)'
11:52:00 | [D] | [p1reader:328] | Could not parse value from line '0-0:1.0.0(230528105200W)'
11:52:00 | [D] | [sensor:093] | 'Cumulative Active Import': Sending state 232.07600 kWh with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Cumulative Active Export': Sending state 1331.51794 kWh with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Cumulative Reactive Import': Sending state 0.86900 kvarh with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Cumulative Reactive Export': Sending state 314.38101 kvarh with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Momentary Active Import': Sending state 0.00000 kW with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Momentary Active Export': Sending state 3.65900 kW with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
11:52:00 | [D] | [sensor:093] | 'Momentary Reactive Export': Sending state 0.86200 kvar with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Import Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Export Phase 1': Sending state 1.06200 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Import Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Export Phase 2': Sending state 1.30900 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Import Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Active Export Phase 3': Sending state 1.28700 kW with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Momentary Reactive Import Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
11:52:01 | [D] | [sensor:093] | 'Current Phase 1': Sending state 4.80000 A with 1 decimals of accuracy
11:52:04 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -72.00000 dBm with 0 decimals of accuracy
11:52:07 | [D] | [switch:012] | 'p1_cts' Turning ON.
11:52:07 | [D] | [switch:055] | 'p1_cts': Sending state ON
11:52:07 | [D] | [switch:012] | 'status_led' Turning ON.
11:52:07 | [D] | [switch:055] | 'status_led': Sending state ON
11:52:07 | [W] | [p1reader:227] | Unknown data format (0x00). Resetting.
11:52:07 | [D] | [switch:016] | 'p1_cts' Turning OFF.
11:52:07 | [D] | [switch:055] | 'p1_cts': Sending state OFF
11:52:07 | [D] | [switch:016] | 'status_led' Turning OFF.
11:52:07 | [D] | [switch:055] | 'status_led': Sending state OFF
11:52:14 | [D] | [sensor:093] |  
``` 
[image]
[image]

Beaky2000, 2023-05-29:
This is a weird one. First I though it would be an unreliable connection causing noise on the serial line, but since the error is always in the same place and with the same "noise" (0x00), that seems unlikely.

My second thought was that the meter might send some extra (0x00) byte(s) after the regular data, but if that was the case, every other attempt to read would be successful, since the error recovery would remove any extra data from the buffer. And if I understand your graphs corerctly, there are long periods of time when you do not get any valid data, right?

One thing you could try, if you want, is to increase the error recovery time. This is a long shot, but if the meter gets into some bad state which it can't get out of because new data is beeing requested too quickly, this might help.
On this line in the code:
else if (500 < loop_start_time - m_error_recovery_time) ChangeState(states::WAITING);
500 is the time in milliseconds to wait and discard any data recevied in case of an error. You could try to increase to say 30000 for 30 seconds and see if it helps. Like I said, it is a long shot.

You could also try and request updates less often and see if the meter behaves better. The default is 2 seconds and can be changed in the yaml file or via a sensor for Home Assistant.

Other than that I think I would want to dump the contents of the buffer after that incorrect 0x00. If I created a branch which did this, would you be able to test it and report back?

doe81, 2023-05-29:
> This is a weird one. First I though it would be an unreliable connection causing noise on the serial line, but since the error is always in the same place and with the same "noise" (0x00), that seems unlikely.
>
Could it be caused by my bad soldering skills? 😶‍🌫️ 
[image]
[image]

> One thing you could try, if you want, is to increase the error recovery time. This is a long shot, but if the meter gets into some bad state which it can't get out of because new data is beeing [...]

I tested the update frequency and it still gets the same error. Also implementing the code above didn't really help more than the 30sec wait when the error showed up.

The stages thing is that I update relatively frequently during times of load. During the daytime when I have my solar plant exporting to the grid or during night time when I'm charging the car. But during times of low load, i get this error all the time.
My first thought was that it was limiting the updates on the P1 port due to no/smal activity. But a friend said that his meter updates all the time.

> Other than that I think I would want to dump the contents of the buffer after that incorrect 0x00. If I created a branch which did this, would you be able to test it and report back?
>

Sure let's try that.

Beaky2000, 2023-05-29:
While I see some room for improvement with your soldering, I doubt that is the problem, but if you can get that tidied up, it won't hurt. Also, how long is the cable? Shorter cables are always better when it comes to avoiding interference.

Does your friend also run a p1mini? On the exact same kind of meter?

I will see when I have time to implement logging of the data discarded in the error recovery state.

Beaky2000, 2023-05-29:
Ok, I have added the logging to the main branch. Update to the latest version and see what the log says efter the error!

Beaky2000, 2023-05-30:
This is getting stranger and stranger... Sometimes when an update is requested, the meter sends an update with correct checksum, but sometimes it sends a few zero bytes...

I also notice that even when correct data is being received, the sending of data is very slow: "Cycle times: Message = 2242 ms (134 loops)". This is over two seconds to transfer a moderate amount of data! I am starting to wonder if the meter is bad, either your particular unit or the design in general.

There is one thing that would be interesting to test, because it could affect several potential issues. It does involve more soldering though: Connect the RTS cable directly to 5V instead of D2.

- This way the RTS signal will be 5V instead of the 3.3V from the ESP output, which is lower than the minimum specified value on many meters. So far I have not heard of any meters where 3.3V does not work, but it is outside of spec.
- The RTS signal will be constantly high and it seems to be part of the P1 protocol that this means that the meter will send updates every 10 seconds. Maybe the meter works better in this mode.
- The RTS signal will not go low as soon as those 0x00 bytes are received and maybe there will be more data giving a clue about what is going on.

Beaky2000, 2023-05-30:
25 cm should not be a problem. I remember one particular meter (can't remember exactly which one) specifying max 3 meters.

doe81, 2023-05-30:
When I have a high load I get some 0 sometimes up to 4. But it reads the meter successfully several times.
Now when there is a very low load I get several 0 even some "e".

Tomorrow I will try to re-soldering.

```
21:33:48 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:33:50 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:33:50 | [W] | [p1reader:218] | Discarding: 000000000000000000
21:33:52 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:33:52 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:33:54 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:33:54 | [W] | [p1reader:218] | Discarding: 00000000000000
21:33:56 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:33:56 | [W] | [p1reader:218] | Discarding: 000000000000
21:33:58 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:33:58 | [W] | [p1reader:218] | Discarding: 00000000000000
21:34:00 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:00 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:34:02 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:02 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:34:04 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:04 | [W] | [p1reader:218] | Discarding: 00000000000000000000
21:34:06 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:06 | [W] | [p1reader:218] | Discarding: 00000000000000
21:34:08 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:08 | [W] | [p1reader:218] | Discarding: 00000000
21:34:10 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:10 | [W] | [p1reader:218] | Discarding: 00000000
21:34:12 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:12 | [W] | [p1reader:218] | Discarding: 0000000000
21:34:14 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:14 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:34:16 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:16 | [W] | [p1reader:218] | Discarding: 00000000000000
21:34:18 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:18 | [W] | [p1reader:218] | Discarding: 000000000000
21:34:20 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:20 | [W] | [p1reader:218] | Discarding: 000000000000000000
21:34:22 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:22 | [W] | [p1reader:218] | Discarding: 000000000000000000
21:34:24 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:24 | [W] | [p1reader:218] | Discarding: 0000000000000000000000
21:34:26 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:26 | [W] | [p1reader:218] | Discarding: 0000000000000000
21:34:28 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:28 | [W] | [p1reader:218] | Discarding: 00000000000000000000
21:34:30 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:30 | [W] | [p1reader:218] | Discarding: 000000e00000
21:34:32 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:32 | [W] | [p1reader:218] | Discarding: 000000000000
21:34:34 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:34 | [W] | [p1reader:218] | Discarding: 000000000000
21:34:36 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
21:34:36 | [W] | [p1reader:218] | Discarding: 000000000000
21:34:38 | [W] | [p1reader:249] | Unknown data format (0x00). Resetting.
```

doe81, 2023-06-01:
I can test the new update no problem.
Thanks @Beaky2000 for your help 👏🏽

Beaky2000, 2023-06-06:
Ok, @doe81, I have created a new branch 'rts-always-high' for you to test and verify. The steps to do this would be:

1. Solder RTS back to D2
2. Switch to the rts-always-high branch and flash
3. Verify that the problems are back
4. Look at the Readme.md (in the branch) for instructions
5. Verify that the problems are solved

doe81, 2023-06-09:
Hi.
I tested, and I got the error message back after re-soldering and flash the rts-always-high branch.
After editing the yaml file according to the readme.md it worked.
```
23:41:38 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:38 | [W] | [p1reader:229] | Discarding: 0000
23:41:39 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:39 | [W] | [p1reader:229] | Discarding: 000000
23:41:40 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:40 | [W] | [p1reader:229] | Discarding: 0000000000
23:41:41 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:41 | [W] | [p1reader:229] | Discarding: 0000
23:41:48 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:48 | [W] | [p1reader:229] | Discarding: 00
23:41:48 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:48 | [W] | [p1reader:229] | Discarding: 0000000000
23:41:49 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:49 | [W] | [p1reader:229] | Discarding: 0000
23:41:50 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:50 | [W] | [p1reader:229] | Discarding: 0000
23:41:51 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:51 | [W] | [p1reader:229] | Discarding: 0000
23:41:52 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:52 | [W] | [p1reader:229] | Discarding: 00
23:41:53 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:53 | [W] | [p1reader:229] | Discarding: 0000
23:41:54 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:54 | [W] | [p1reader:229] | Discarding: 0000
23:41:55 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:55 | [W] | [p1reader:229] | Discarding: 00000000
23:41:56 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:41:57 | [W] | [p1reader:258] | Unknown data format (0x00). Resetting.
23:42:04 | [I] | [ota:349] | OTA update finished!
23:42:04 | [I] | [app:134] | Rebooting safely...
23:45:28 | [W] | [api.connection:087] | ESPHome Logs 2023.5.5 (192.168.68.104): Connection closed
23:47:15 | [I] | [ota:113] | Boot seems successful, resetting boot loop counter.
``` 

👍👏🏽🔝

Beaky2000, 2023-06-09:
Excellent! Then I will merge this branch into main in a couple of days!

closed 2023-06-09 by Beaky2000