Beaky2000/esphome-p1mini #14: P1 mini disconnecting

CrallH, 2023-10-04:
Logger: aioesphomeapi.connection
Source: runner.py:186
First occurred: October 3, 2023 at 11:09:48 (491 occurrences)
Last logged: 13:02:22

p1mini @ 192.168.99.242: Connection error occurred: [Errno 104] Connection reset by peer
p1mini @ 192.168.99.242: Connection error occurred: p1mini @ 192.168.99.242: Connection lost
p1mini @ 192.168.99.242: Connection error occurred: p1mini @ 192.168.99.242: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (Home Assistant 2023.9.3).

Beaky2000, 2023-10-04:
I don't know why that is happening and it is possible that it is an ESPHome issue, rather than a p1mini issue.

I do notice that the WiFi signal is fairly low at -81 db, so it would be interesting to see if the problem is fixed if the signal is better (by putting a repeater between p1mini and access point etc).

CrallH, 2023-10-04:
There is a unifi Access Point 2m from the unit  (its in metal box thou, fasadmätarskåp). Can see if I can make a longer cabel and/or get it thru the wall.

CrallH, 2023-10-04:
Just open the door gave me -52bB. So I need to figure out how to move it. 
This droput happens alot, but unifi says it's online all the time, so it's not disconnect from wifi.

CrallH, 2023-10-04:
Geting the same message at -56dB
[image]
[image]

Beaky2000, 2023-10-04:
Ok, so it is not the network connection then.

I have updated my p1mini to the latest ESPHome version (2023.9.3) but I am running an older Home Assistant (2023.4.6) with no problems. I will update my Home Assistant when I have some time to spare, in case there is some problem with later versions.

I would also suggest you update ESPHome to the latest version for the same reason.

There is also a suggestion in your error messages that sound like it would be worth trying, to see if it makes a difference: "Try enabling encryption on the device or turning off encryption on the client"

Beaky2000, 2023-10-04:
Did some Googling and found this thread: https://github.com/esphome/issues/issues/3108

This seems to have helped some people:
> > Sure - in HA i went to Configuration --> Devices and Services --> under the ESPHome tile I had a previous configuration used for the same board. I then chose that device -> 3 dots menu --> Delete.
> 
> Now that device is erased from HA, and so the MAC address of this board is not longer associated to an existing device -> it's recognized as new device. All errors previously shown in the HA logs no longer appear, and the logs of the board itself (viewing through ESPHome) are clean as well - no disconnections.
> Hope this helps.

CrallH, 2023-10-04:
I'm updating to everything new now during the release stream 😊
Will test the linked proposal tomorrow 👍

CrallH, 2023-10-04:
Can't even update the mini without it resetting the connection.....

CrallH, 2023-10-04:
Updated everything to latest version, eaphome, home Assistant and p1 mini. Deleted it in integrations, added again, still same issue.
Will test a second D1 mini tomorrow.

CrallH, 2023-10-25:
We can close this, I finally got some time to exchange my D1 MINI (NON clone) to a new D1 mini (clone). Even with lower RSSI it's not dropping out like the first one did.

closed 2023-10-25 by CrallH