Beaky2000/esphome-p1mini #16: Not receiving sensor updates after plugging into power meter.

FreshJonte, 2023-10-14:
Hello!

I found this interesting project in a swedish facebook group. Immediately got a RJ12 cable and found a D1 mini lying around and soldered it together.

After connecting the D1 mini to my EON powermeter i immediately get a sensor reading but after that no more updates. The only sensor that updates is the wifi sensor. I used the EON elna before so i know that the HAN port is open and working.

Do you have any idea why it is not working for me? Attaching some info. Tell me if you need anything else.

Thanks!

Regards / med vänlig hälsning
Jonatan

[logs_p1mini_logs.txt](https://github.com/Beaky2000/esphome-p1mini/files/12906246/logs_p1mini_logs.txt)
[image]

Beaky2000, 2023-10-14:
So are you getting a single reading from the meter on startup? 13.6 A sounds like a lot of current, does that seem correct?

What kind of meter do you have? EON seems to be using Landis+Gyr E360 a lot, so is that it?

I would suggest double checking all the connections to make dure they are correct. I guess ground and 5V are correct or you wouldn't get anything, but check the other connections.

It could also be interesting to test in the RTS always high mode: https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md

FreshJonte, 2023-10-14:
Yes that is correct, 1 reading after plugging it in. I agree that it is a bit high but momentarily that could be correct.

Yes that is correct, Landis+Gyr E360.

I have double checked the connections and they seem correct to me. However my cable is about 15-20cm longer than in your pictues. Could that be a problem?

I tried the RTS thing before and it didn't work. Just tried it for the second time but no change. Now after uploading new file all sensors are unknown instead. So it does not seem to recieve any data.

FreshJonte, 2023-10-15:
Okay so I cut down the cable and resoldered everything but no difference. Guess next step is to try new D1 mini?

Beaky2000, 2023-10-15:
I am pretty sure that no data at all is being received on the serial line. I am considering adding some log output in cases like this. Not that it would solve anything, but it might help with debugging.

There could be several reasons no data is received and ideally, the first step would be to verify that the meter is actually outputting something. I would have measured the output with an oscilloscope to see if there is any activity, but I realize that most people don't have access to one...

Since the output is open drain, you could diconnect the TX signal from the D1mini and connect an LED in series with a resistor (say 1.5k to 3k) from 5V to TX. If the LED does not light at all there is nothing being sent on the output!

You could double check that the port is still active. My (Sagemcom) meter has an indicator in the display but I am not sure if the E360 has something similar? If not check with EON I guess.

Also, it seems like "older" E360 meters have a bug which causes them to stop sending on the port in some cases and only start working again after a complete power cycle. If this turns out to be the case I guess see if EON can update the firmware in the meter.

But it could also be a problem with the D1mini, so trying a different one is not a bad idea. In case the problem is with the internal pullup resistor, you could try connecting a resistor between RX and 3.3V on the D1mini (1k - 5k should be fine).

FreshJonte, 2023-10-15:
Thank you for your suggestions! I did some testing that could hopefully help narrow it down. Im reading on this swedish forum: https://www.byggahus.se/forum/threads/elmaetare-h1-port-p1-port-han-port-manga-namn-verkar-florera.384608/page-36
That it could be a power issue with the D1. So i connected the USB on D1 mini to my laptop and reset but that didnt seem to work. 
Then I tried measuring with my multimeter between RX and GND on the D1 and that seems to give me a constant voltage of 0,257volts and suddenly I seemed to get a reading! But unfortunately no more than one reading. According to that forum the voltage should come every 10 seconds but i am getting a constant voltage, that might be weird? But atleast the output is giving something and that is good i guess.

I have attached logs hopefully they can help you narrow it down. Thanks for your patience and help :)

[logs_p1mini_logs.txt](https://github.com/Beaky2000/esphome-p1mini/files/12910136/logs_p1mini_logs.txt)

Beaky2000, 2023-10-15:
That is both weird and interesting... One thing you could play about with is to remove the internal pullup resistor on RX (change INPUT_PULLUP to INPUT in the yaml) and try an external pullup resistors between 3.3V and RX. Maybe try values like 1k, 5k, 10k, etc and see if anything works...

Also, it wouldn't hurt to try a different D1mini, in case there is some issue with the one you have.

FreshJonte, 2023-11-03:
Hello again. As you probably suspected I gave up on getting this to work. I ordered a SliemmLezer+, plugged it in today and that was working perfectly. But thank you, I appreciate your support :)

closed 2023-11-03 by FreshJonte