Beaky2000/esphome-p1mini #22: No reading on E360

dehooge, 2024-01-31:
Hey, I just tried this but I get no data from meter. Landis E360. Han port is active and working with another meter.
Rewired twice and tried two different D1.

Logs give me this:
[13:51:11][W][p1reader:251]: No data received for 60 seconds.
[13:51:11][D][switch:016]: 'p1_cts' Turning OFF.
[13:51:11][D][switch:055]: 'p1_cts': Sending state OFF
[13:51:12][D][switch:016]: 'status_led' Turning OFF.
[13:51:12][D][switch:055]: 'status_led': Sending state OFF
[13:51:12][D][switch:012]: 'p1_cts' Turning ON.
[13:51:12][D][switch:055]: 'p1_cts': Sending state ON
[13:51:12][D][switch:012]: 'status_led' Turning ON.
[13:51:12][D][switch:055]: 'status_led': Sending state ON

I tested the advice given in this [issue ](https://github.com/Beaky2000/esphome-p1mini/issues/16) and set RTS to always high and put a 10k resistor between 3.3 and RX, no luck.

Any more suggestions for me?

dehooge, 2024-02-01:
I don't have an oscilloscope unfortunately and removing the resistors didn't change anything.

It's working with the Tibber P1 meter, wonder what they can be doing differently to get data.

Beaky2000, 2024-02-01:
It would be interesting to see what is different with the Tibber meter. Unfortunately I can't think of a way of doing that that is not pretty technical, such as attaching an oscilloscope...

dehooge, 2024-02-01:
Guess I'll be ordering an oscilloscope, always wanted an excuse...
Gonna crack open the Tibber meter and se what's going on there as well, think the run an esp as well.

dehooge, 2024-02-06:
Found the problem. Needed to feed RTS with 5v. Tried with 3.3V but guess this wasn't enough.

Beaky2000, 2024-02-06:
That is very interesting. 5V is according to spec, but previously I was not aware of any meter that did not work with 3.3V.

I assume that now you just connected RTS directly to 5V, right?

If you have the time, it would be very interesting to see if RTS could still be controlled from a GPIO by doing this:

* Set the output pin to OUTPUT_OPEN_DRAIN
* Connect an external pull-up resistor, maybe 2k - 5k somewhere, between the pin and 5V.
* Connect the pin to RTS on the meter, like before.

If we can verify this as a solution, I can add it to the documentation.

dehooge, 2024-02-06:
Yes I just connected RTS to 5V.

Do you mean for me to connect a resistor between 5v and the D2 pin?

Beaky2000, 2024-02-06:
Yes, exactly. But configure the pin as "open drain" first. We don't want it to output 3.3V while connected via the resistor to 5V!

```
  - platform: gpio
    id: p1_cts
    pin:
      number: D2
      mode: OUTPUT_OPEN_DRAIN
```

dehooge, 2024-02-06:
Tried this but got no reading unfortunately.

Beaky2000, 2024-02-06:
That is a shame. What value resistor did you use? It could be a case of needing a lower resistance for it to work. Figuring out what value is just right is another case where having that oscilloscope would be really nice! :)

If you don't want to keep messing with this I understand. I will add some info and suggestions to the Readme now that I know that this can be an issue!

closed 2024-02-06 by Beaky2000

dehooge, 2024-02-06:
2k resistor. I don't mind messing with it but still haven't order the oscilloscope ;)

Beaky2000, 2024-02-06:
Ok, 2k seems fairly low. It could be that the ESP is preventing the pin from going high enough even in open-drain mode. I know I have had problems using open-drain with 12V signals connected to an ESP, but 5V seems to have worked for me in the past.

Without "seeing" what is happening on the pin, there is going to be a lot of guessing...

gunstr, 2024-02-06:
I have checked the same in the past for other applications and as per the spec max voltage for any IO is Vdd + 0.3V. It's not clearly mentioned in the documentation but it's likely there are some clamping mechanism to rail that probably kicks in and will limit the voltage to an unspecified level also in open drain mode. I ended up always using level shifters...

Beaky2000, 2024-02-06:
That sounds like a probable explanation!

A transistor and two resistors should work (or a dedicated level shifter if that is simpler), but I am not sure it would be worth the effort compared to just setting RTS high all the time. I am not sure how often the E360 updates when RTS is constantly high or how often it can update when prompted by the RTS signal.

Fywqdk, 2024-07-03:
Hi sorry for reviving this. I have a Landis+Gyr 360 as well and the same problem as in OP. First off I was unable to run my board (Lolin NodeMCU V3) on the port alone so connected an old USB phone charger. Then I get connection by WIFI and can do OTA updates etc. but there's no data. Tried pulling the RTS to 5V using the boards V USB but that didn't do anything. I even tried the LED and resistor suggestion from the other issue listed in OP but nothing at all in terms of data, and if I disconnect the power from the USB side it just dies and cant power on. I am starting to suspect my power company hasn't switched on the port as promised, but I do seem to get something out of it.

I am not that strong in this area, so any pointers or ideas would be welcome.