Beaky2000/esphome-p1mini #24: Unknown data format.

rulabmarton, 2024-02-04:
I tried to update my working self made P1 meter.

Hardware: WemosD1 Mini Pro/S34U18 (Sanxing SX631) 
Custom component: https://github.com/psvanstrom/esphome-p1reader (Wiring wemos D1 mini)

Log output:
01:26:39 | [W] | [P1Mini:127] | Unknown data format (0x68). Resetting.
01:26:39 | [W] | [P1Mini:435] | Discarding: 5f558a191919c6468633336686460c8c196619793d793d461ac68cc6e3863486

what am I missing ?

Beaky2000, 2024-02-04:
If you are inverting the signal with the transistor, you should not invert
it in the config file, so you probably need to change “invert: true” to
“invert: false” on RX.

rulabmarton, 2024-02-04:
I really appreciate your help.
I corrected it, but now: [15:47:40][W][P1Mini:177]: Message buffer overrun. Resetting.
[15:47:40][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[15:47:40][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
I also tried to change INPUT_PULLUP to INPUT
....
It's not an hardware issue, because it works well with tasmota.

Beaky2000, 2024-02-04:
That is weird, but using INPUT instead of INPUT_PULLUP is correct when inverting with the transistor. You should also set "minimum_period: 0s" (since I assume you do not have RTS attached to a GPIO).  But that does not explain the issue you have now. Does it work better with the version in the main branch or is that the same?

rulabmarton, 2024-02-04:
I tried the main branch, but it not working for me, Esphome ver. 2023.12.9.

**Log output:**
```
In file included from src/main.cpp:70:
src/p1mini.h:1:1: error: expected unqualified-id before '{' token
...
      | ^
<unicode string>: In lambda function:
<unicode string>:101:31: error: expected type-specifier before 'P1Reader'
<unicode string>:136:7: error: could not convert '{<expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
*** [.pioenvs/p1mini/src/main.cpp.o] Error 1
```

Beaky2000, 2024-02-20:
Since you have the SX631, you might be having the same issue as this: https://github.com/Beaky2000/esphome-p1mini/issues/25

Simply increasing the internal buffer size might be all that is needed.

rulabmarton, 2024-02-22:
The solution works with SX631 and P1mini pro (hardware inverted logic)

closed 2024-02-22 by rulabmarton