Beaky2000/esphome-p1mini #25: error reading data

CadilLACi, 2024-02-15:
Hello!

I am getting a strange error, every 10 seconds, it writes this:

[22:43:07][W][P1Mini:177]: Message buffer overrun. Resetting.
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
[22:43:07][W][P1Mini:435]: Discarding: ffff290d0a21353846320d0affffffffffffffffffffffffffffffffffffffff

I have followed the inscrutcions, and I have the p1mini connected. It boots from the Electricity Meter's power. It has it's website online, it connects to the homeassistant instance, but i get no data. 

I have an sx631 meter, so i removed the who lines that switch.turn_off: p1_rts 

What did go wrong?

thx in advance

CadilLACi

Beaky2000, 2024-02-16:
The SX631 seems to be "difficult" and possibly buggy in general... That 290d0a2... at the end is the end of the actual data, so there is nothing wrong with the connection. The meter just seems to take longa pauses, while sometimes spamming a lot of noise.

If your cable is long, say more than a couple of decimeters, I would suggest trying a shorter cable.

This (copied from another message) could possibly be relevant:

```
When I was trying to get a D1mini to communicate with the module in an Ikea Vindriktning over the UART, I could not get it to work. The baud rate was much lower, so I could use soft serial on other pins, but the hardware UART would not work.

I eventually figured out that since the UART is connected to the USB communication chip (via two resistors) and even though the USB chip was not active, that caused enough interference to disrupt the serial communication. By removing the resistors (which disables USB completely) I was able to get it to work.

This could be similar to what is happening in your case. The signal from the meter could be correct, but not "strong" enough to share the connection with the USB chip.
```

But other than that I have no idea. The message must have started correct since if the first byte is incorrect there is a different error and the end of the message is there, so there is just a section in the middle with a lot of "ff".

Beaky2000, 2024-02-16:
This is pretty wild and crazy, but since the "noise" in your case always seems to be "ff", you could try this as a workaround:

In p1mini.cpp, find this code:

```
                    // While data is available, read it one byte at a time.
                    char const read_byte{ GetByte() };
```
and add one extra line:
```
                    // While data is available, read it one byte at a time.
                    char const read_byte{ GetByte() };
                    if (read_byte == 0xff) continue;
```

It would be interesting to see if this works!

CadilLACi, 2024-02-16:
Hi!

Now I get these messages:

23:55:31 | [W] | [P1Mini:219] | CRC mismatch, calculated A0CB != 2C3F. Message ignored.
23:55:38 | [W] | [P1Mini:219] | CRC mismatch, calculated 605B != 2CBE. Message ignored.
23:55:48 | [W] | [P1Mini:219] | CRC mismatch, calculated 3CF6 != 86BF. Message ignored.
23:55:58 | [W] | [P1Mini:219] | CRC mismatch, calculated AD99 != 0FCE. Message ignored.

Does this help?

CadilLACi, 2024-02-16:
I have an 80cm cable. I cannot storten it, because the meter is in such a box that I have to snake the cable around to get to it. I cannot simply open it and connect to it directly, the cover is wired with a signetlike seal.

Beaky2000, 2024-02-18:
Don't worry about the cable lengt. I do not think that is the problem.

It is a shame that the ignore-ff-hack did not work. I have been thinking about dumping the entire buffer in hex to the log so it could be examined in detail, but that wouldn't really solve your issue. Possibly work around it though...

I would be interested to see if removing those resistors attaching the UART to the USB chip would help. They are usually not that difficult to locate and can usually be removed with brute force. But that will disable all USB connectivity, so you will be stuck with OTA only updates. It is up to you if you want to try it!

CadilLACi, 2024-02-19:
Hmmm, I turned the debug level from info to debug adn added some lines to the uart part in the yaml to debug:

  debug:
    direction: BOTH
    dummy_receiver: false  

With that, the log expanded to this:

15:23:05	[D]	[sensor:093]	
'p1mini WiFi Signal': Sending state -68.00000 dBm with 0 decimals of accuracy
15:23:06	[D]	[P1Mini:119]	
ASCII data format
15:23:06	[D]	[uart_debug:114]	
<<< 2F:41:55:58:35:39:39:30:32:37:33:39:39:34:33:0D:0A:0D:0A:30:2D:30:3A:31:2E:30:2E:30:28:32:34:30:32:31:39:31:35:32:33:31:30:57:29:0D:0A:30:2D:30:3A:34:32:2E:30:2E:30:28:41:55:58:31:30:33:30:33:30:32:37:33:39:39:34:33:29:0D:0A:30:2D:30:3A:39:36:2E:31:2E:30:28:39:39:30:32:37:33:39:39:34:33:29:0D:0A:30:2D:30:3A:39:36:2E:31:34:2E:30:28:30:30:30:31:29:0D:0A:30:2D:30:3A:39:36:2E:35:30:2E:36:38:28:4F:4E:29:0D:0A:30:2D:30:3A:31:37:2E:30:2E:30:28:39:30:2E
15:23:06	[D]	[uart_debug:114]	
<<< 30:30:30:2A:6B:57:29:0D:0A:31:2D:30:3A:31:2E:38:2E:30:28:30:31:33:33:33:38:2E:30:36:34:2A:6B:57:68:29:0D:0A:31:2D:30:3A:31:2E:38:2E:31:28:30:30:35:31:38:30:2E:34:33:38:2A:6B:57:68:29:0D:0A:31:2D:30:3A:31:2E:38:2E:32:28:30:30:38:31:35:37:2E:36:32:36:2A:6B:57:68:29:0D:0A:31:2D:30:3A:31:2E:38:2E:33:28:30:30:30:30:30:30:2E:30:30:30:2A:6B:57:68:29:0D:0A:31:2D:30:3A:31:2E:38:2E:34:28:30:30:30:30:30:30:2E:30:30:30:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E
15:23:06	[D]	[uart_debug:114]	
<<< 38:2E:30:28:30:31:30:30:34:34:2E:38:36:36:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:31:28:30:30:37:31:39:33:2E:35:31:37:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:32:28:30:30:32:38:35:31:2E:33:34:39:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:33:28:30:30:30:30:30:30:2E:30:30:30:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:34:28:30:30:30:30:30:30:2E:30:30:30:2A:6B:57:68:29:0D:0A:31:2D:30:3A:33:2E:38:2E:30:28:30:30:30:37:32:33:2E:39:35:34:2A
15:23:06	[D]	[uart_debug:114]	
<<< 6B:76:61:72:68:29:0D:0A:31:2D:30:3A:34:2E:38:2E:30:28:30:30:34:31:36:36:2E:37:30:35:2A:6B:76:61:72:68:29:0D:0A:31:2D:30:3A:35:2E:38:2E:30:28:30:30:30:34:32:31:2E:36:35:32:2A:6B:76:61:72:68:29:0D:0A:31:2D:30:3A:36:2E:38:2E:30:28:30:30:30:33:30:32:2E:33:30:32:2A:6B:76:61:72:68:29:0D:0A:31:2D:30:3A:37:2E:38:2E:30:28:30:30:30:36:34:30:2E:32:39:30:2A:6B:76:61:72:68:29:0D:0A:31:2D:30:3A:38:2E:38:2E:30:28:30:30:33:35:32:36:2E:34:31:35:2A:6B:76:61:72:68
15:23:06	[D]	[uart_debug:114]	
<<< 29:0D:0A:31:2D:30:3A:31:35:2E:38:2E:30:28:30:32:33:33:38:32:2E:39:34:30:2A:6B:57:68:29:0D:0A:31:2D:30:3A:33:32:2E:37:2E:30:28:32:33:36:2E:30:2A:56:29:0D:0A:31:2D:30:3A:35:32:2E:37:2E:30:28:32:33:35:2E:34:2A:56:29:0D:0A:31:2D:30:3A:37:32:2E:37:2E:30:28:32:33:39:2E:36:2A:56:29:0D:0A:31:2D:30:3A:33:31:2E:37:2E:30:28:30:30:32:2A:41:29:0D:0A:31:2D:30:3A:35:31:2E:37:2E:30:28:30:30:30:2A:41:29:0D:0A:31:2D:30:3A:37:31:2E:37:2E:30:28:30:30:30:2A:41:29:0D
15:23:06	[D]	[uart_debug:114]	
<<< 0A:31:2D:30:3A:31:33:2E:37:2E:30:28:30:2E:37:37:39:29:0D:0A:31:2D:30:3A:33:33:2E:37:2E:30:28:30:2E:37:37:36:29:0D:0A:31:2D:30:3A:35:33:2E:37:2E:30:28:30:2E:38:30:38:29:0D:0A:31:2D:30:3A:37:33:2E:37:2E:30:28:30:2E:30:39:33:29:0D:0A:31:2D:30:3A:31:34:2E:37:2E:30:28:34:39:2E:39:39:2A:48:7A:29:0D:0A:31:2D:30:3A:31:2E:37:2E:30:28:30:30:2E:33:39:34:2A:6B:57:29:0D:0A:31:2D:30:3A:32:2E:37:2E:30:28:30:30:2E:30:30:30:2A:6B:57:29:0D:0A:31:2D:30:3A:35:2E:37
15:23:06	[D]	[uart_debug:114]	
<<< 2E:30:28:30:30:2E:30:31:35:2A:6B:76:61:72:29:0D:0A:31:2D:30:3A:36:2E:37:2E:30:28:30:30:2E:30:30:30:2A:6B:76:61:72:29:0D:0A:31:2D:30:3A:37:2E:37:2E:30:28:30:30:2E:30:32:39:2A:6B:76:61:72:29:0D:0A:31:2D:30:3A:38:2E:37:2E:30:28:30:30:2E:33:30:32:2A:6B:76:61:72:29:0D:0A:30:2D:30:3A:39:38:2E:31:2E:30:28:32:34:30:32:30:31:30:30:30:30:30:30:57:29:28:30:31:32:38:33:38:2E:39:35:37:2A:6B:57:68:29:28:30:30:35:30:31:31:2E:36:30:32:2A:6B:57:68:29:28:30:30:37
15:23:06	[W]	[component:214]	
Component p1_mini took a long time for an operation (0.16 s).
15:23:06	[W]	[component:215]	
Components should block for at most 20-30ms.
15:23:07	[D]	[uart_debug:114]	
<<< 38:32:37:2E:33:35:35:2A:6B:57:68:29:28:30:30:39:38:36:31:2E:35:37:36:2A:6B:57:68:29:28:30:30:37:30:35:39:2E:33:38:33:2A:6B:57:68:29:28:30:30:32:38:30:32:2E:31:39:33:2A:6B:57:68:29:28:30:30:30:37:31:35:2E:37:30:32:2A:6B:76:61:72:68:29:28:30:30:33:39:38:39:2E:39:32:31:2A:6B:76:61:72:68:29:28:30:30:30:34:31:37:2E:36:38:30:2A:6B:76:61:72:68:29:28:30:30:30:32:39:38:2E:30:32:32:2A:6B:76:61:72:68:29:28:30:30:30:36:31:37:2E:31:39:35:2A:6B:76:61:72:68:29
15:23:07	[D]	[uart_debug:114]	
<<< 28:30:30:33:33:37:32:2E:37:32:36:2A:6B:76:61:72:68:29:28:30:32:32:37:30:30:2E:35:34:32:2A:6B:57:68:29:28:30:35:2E:35:30:30:2A:6B:57:29:28:30:35:2E:35:30:30:2A:6B:57:29:28:30:33:2E:39:32:30:2A:6B:57:29:28:30:33:2E:33:30:30:2A:6B:57:29:28:30:33:2E:33:30:30:2A:6B:57:29:28:30:33:2E:31:35:36:2A:6B:57:29:0D:0A
15:23:07	[D]	[uart_debug:114]	
<<< 30:2D:30:3A:39:36:2E:31:33:2E:30:28:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
15:23:07	[D]	[uart_debug:114]	
<<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:F [... cut ...]
<<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:29:0D:0A:21:32:30:43:34:0D:0A
15:23:07	[D]	[switch:012]	
'p1_rts' Turning ON.
15:23:07	[D]	[switch:012]	
'status_led' Turning ON.
15:23:07	[D]	[switch:055]	
'status_led': Sending state ON

Is this helpful somehow?

Beaky2000, 2024-02-19:
If I run the hex codes, without all the FFs, through a Hex to ascii converter (https://www.rapidtables.com/convert/number/hex-to-ascii.html) I get this:

```
/AUX59902739943

0-0:1.0.0(240219152310W)
0-0:42.0.0(AUX1030302739943)
0-0:96.1.0(9902739943)
0-0:96.14.0(0001)
0-0:96.50.68(ON)
0-0:17.0.0(90.000*kW)
1-0:1.8.0(013338.064*kWh)
1-0:1.8.1(005180.438*kWh)
1-0:1.8.2(008157.626*kWh)
1-0:1.8.3(000000.000*kWh)
1-0:1.8.4(000000.000*kWh)
1-0:2.8.0(010044.866*kWh)
1-0:2.8.1(007193.517*kWh)
1-0:2.8.2(002851.349*kWh)
1-0:2.8.3(000000.000*kWh)
1-0:2.8.4(000000.000*kWh)
1-0:3.8.0(000723.954*kvarh)
1-0:4.8.0(004166.705*kvarh)
1-0:5.8.0(000421.652*kvarh)
1-0:6.8.0(000302.302*kvarh)
1-0:7.8.0(000640.290*kvarh)
1-0:8.8.0(003526.415*kvarh)
1-0:15.8.0(023382.940*kWh)
1-0:32.7.0(236.0*V)
1-0:52.7.0(235.4*V)
1-0:72.7.0(239.6*V)
1-0:31.7.0(002*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:13.7.0(0.779)
1-0:33.7.0(0.776)
1-0:53.7.0(0.808)
1-0:73.7.0(0.093)
1-0:14.7.0(49.99*Hz)
1-0:1.7.0(00.394*kW)
1-0:2.7.0(00.000*kW)
1-0:5.7.0(00.015*kvar)
1-0:6.7.0(00.000*kvar)
1-0:7.7.0(00.029*kvar)
1-0:8.7.0(00.302*kvar)
0-0:98.1.0(240201000000W)(012838.957*kWh)(005011.602*kWh)(007827.355*kWh)(009861.576*kWh)(007059.383*kWh)(002802.193*kWh)(000715.702*kvarh)(003989.921*kvarh)(000417.680*kvarh)(000298.022*kvarh)(000617.195*kvarh)(003372.726*kvarh)(022700.542*kWh)(05.500*kW)(05.500*kW)(03.920*kW)(03.300*kW)(03.300*kW)(03.156*kW)
0-0:96.13.0()
!20C4
```

which looks like a complete message but with some extra data that I am not used to seeing. Which shouldn't be a problem as long as the checksum is correct, but obviously it is not...

I wonder if adding all the FFs to the checksum calculation would make it match... This is _very_ far fetched, but I want to try it... I am going to have to do some coding thoug, so I am not sure when it will happen.

Beaky2000, 2024-02-19:
I just realized that the easiest way to test that theory is probably for you to increase the buffer size until you no longer get buffer overruns and see what happens.

In p1_mini.h, find this line:

`constexpr static int message_buffer_size{ 2048 };
`

and increase it until you get something other than "buffer overrun". Maybe start with 4096 and see if it is enough. Making the buffer too big might prevent building or running though.

Beaky2000, 2024-02-19:
Ok, newsflash!

I couldn't help myself, so I wrote some code to test the checksum calculations on the data and, believe it or not, all those FFs are included in the checksum!

I assumed all those FFs must be some sort of communication error, probably hardware related! But no, the meter is actively sending those FFs and including them in the checksum calculation!

I am going to need some time to digest this, but I should be able to get it working!

My suggested increase-the-buffer hack is probably going to work reasonably well in the mean time, so do try it out!

Beaky2000, 2024-02-19:
Update: All those FFs are 1024 bytes of data associated with the OBIS tag 0-0:96.13.0, which is described as "Text message max
1024 characters.". Essentailly the meter is sending the letter "ÿ" 1024 times as a text message.

Given that the size of this data is consistent, it should be enough to increase the buffer by around 1024 bytes, so try setting it to 3072 and there is a good chance everything is going to work! (And remove the previous skip-ff-hack)

CadilLACi, 2024-02-19:
Wow, thx for the brainstorm!

So, I did the following:
1. I removed the line from p1_mini.cpp that said: if (read_byte == 0xff) continue; and this fixed the crc errors
2. restored the original state of the p1mini.yaml without the debug informations
3. in the p1_mini.h, I increased the constexpr static int message_buffer_size{ 2048 };  to constexpr static int message_buffer_size{ 3072 };

Now it works!!!!

Thank You!

CadilLACi, 2024-02-19:
It works now!

closed 2024-02-19 by CadilLACi

Beaky2000, 2024-02-19:
Excellent! I think I will add an option to set the buffer size via the yaml and probably set it to 3072 by default.