Beaky2000/esphome-p1mini #26: Shenzhen Star Instrument Discarding

riborg, 2024-03-03:
I get this from my newly installed meter from Swemet and Shenzhen Star Instruments.. I have activated port with my network owner..

Did I wire something wrong?

[22:54:26][I][app:102]: ESPHome version 2024.2.1 compiled on Mar  3 2024, 22:53:31
[22:54:27][W][P1Mini:448]: Discarding: 30302a6b5768290d0a21413931410d0a32343430315729283134323339392e30
[22:54:27][W][component:214]: Component api took a long time for an operation (0.29 s).
[22:54:27][W][component:215]: Components should block for at most 20-30ms.
[22:54:28][W][P1Mini:139]: Unknown data format (0xB9). Resetting.
[22:54:28][W][P1Mini:448]: Discarding: ea75eacd5c324d535a333134580d0a0d0a312d333a302e322e38283530290d0a
[22:54:28][W][P1Mini:448]: Discarding: 302d303a312e302e302832343033303332323534323557290d0a302d303a3936
[22:54:28][W][P1Mini:448]: Discarding: 2e312e3128333733333335333933393339333233353334333333303330333533
[22:54:28][W][P1Mini:448]: Discarding: 373338333033363336290d0a312d303a312e382e31283030343733302e383036
[22:54:28][W][P1Mini:448]: Discarding: 2a6b5768290d0a312d303a312e382e32283030303030302e3030302a6b576829
[22:54:28][W][P1Mini:448]: Discarding: 0d0a312d303a322e382e31283030303030302e3030302a6b5768290d0a312d30
[22:54:28][W][P1Mini:448]: Discarding: 3a322e382e32283030303030302e3030302a6b5768290d0a302d303a39362e31
[22:54:28][W][P1Mini:448]: Discarding: 342e302830303031290d0a312d303a312e382e30283030343733302e3830362a
[22:54:28][W][P1Mini:448]: Discarding: 6b5768290d0a312d303a322e382e30283030303030302e3030302a6b5768290d
[22:54:28][W][P1Mini:448]: Discarding: 0a312d303a332e382e30283030303030302e3531362a6b76617268290d0a312d
[22:54:28][W][P1Mini:448]: Discarding: 303a342e382e30283030313738322e3032392a6b76617268290d0a312d303a31
[22:54:28][W][P1Mini:448]: Discarding: 2e372e302830312e3037362a6b57290d0a312d303a322e372e302830302e3030
[22:54:28][W][P1Mini:448]: Discarding: 302a6b57290d0a312d303a332e372e302830302e3030302a6b766172290d0a31
[22:54:28][W][P1Mini:448]: Discarding: 2d303a342e372e302830302e3438332a6b766172290d0a312d303a39392e3937
[22:54:28][W][P1Mini:448]: Discarding: 2e302831302928302d303a39362e31312e363329283233313230363130323334
[22:54:28][W][P1Mini:448]: Discarding: 3357292836292832333132303631303233343357292835292832333132303631
[22:54:28][W][P1Mini:448]: Discarding: 3032333433572928342928323331323036313032303037572928332928323331
[22:54:28][W][P1Mini:448]: Discarding: 3230363130323030375729283229283233313230363130323030375729283129
[22:54:28][W][P1Mini:448]: Discarding: 2832333130313630373235343557292836292832333130313630373557292834
[22:54:28][W][P1Mini:448]: Discarding: 2e33322e30283030303030290d0a312d303a35322e33322e3028303030303029
[22:54:28][W][P1Mini:448]: Discarding: 0d0a312d303a37322e33322e30283030303030290d0a312d303a33322e33362e
[22:54:28][W][P1Mini:448]: Discarding: 30283030303030290d0a312d303a35322e33362e30283030303030290d0a312d
[22:54:28][W][P1Mini:448]: Discarding: 303a37322e33362e30283030303030290d0a302d303a39362e31332e30283330
[22:54:28][W][P1Mini:448]: Discarding: 333033303330333033303330333033303330333033303330333033303330290d
[22:54:28][W][P1Mini:448]: Discarding: 0a312d303a33322e372e30283233362e382a56290d0a312d303a35322e372e30
[22:54:28][W][P1Mini:448]: Discarding: 283233362e342a56290d0a312d303a37322e372e30283233352e352a56290d0a
[22:54:28][W][P1Mini:448]: Discarding: 312d303a33312e372e30283030302e342a41290d0a312d303a35312e372e3028
[22:54:28][W][P1Mini:448]: Discarding: 3030312e392a41290d0a312d303a37312e372e30283030332e322a41290d0a31
[22:54:28][W][P1Mini:448]: Discarding: 2d303a32312e372e302830302e3035342a6b57290d0a312d303a34312e372e30
[22:54:28][W][P1Mini:448]: Discarding: 2830302e3436302a6b57290d0a312d303a36312e372e302830302e3536312a6b
[22:54:28][W][P1Mini:448]: Discarding: 57290d0a312d303a32322e372e302830302e3030302a6b57290d0a312d303a34
[22:54:28][W][P1Mini:448]: Discarding: 322e372e302830302e3030302a6b57290d0a312d303a36322e372e302830302e
[22:54:28][W][P1Mini:448]: Discarding: 3030302a6b57290d0a312d303a32332e372e302830302e3030302a6b76617229
[22:54:28][W][P1Mini:448]: Discarding: 0d0a312d303a34332e372e302830302e3030302a6b766172290d0a312d303a36
[22:54:28][W][P1Mini:448]: Discarding: 332e372e302830302e3030302a6b766172290d0a312d303a32342e372e302830
[22:54:28][W][P1Mini:448]: Discarding: 302e3035362a6b766172290d0a312d303a34342e372e302830302e3130322a6b
[22:54:28][W][P1Mini:448]: Discarding: 766172290d0a312d303a36342e372e302830302e3332342a6b766172290d0a30
[22:54:28][W][P1Mini:448]: Discarding: 2d313a32342e312e3028303034290d0a302d313a39362e312e30283336333833
[22:54:28][W][P1Mini:448]: Discarding: 3933333339333833373337290d0a302d313a32342e322e312832343033303332
[22:54:28][W][P1Mini:448]: Discarding: 32343430315729283134323339392e3030302a6b5768290d0a21463243380d0a

Beaky2000, 2024-03-04:
If I put that data into [a hex to ascii converter](https://www.rapidtables.com/convert/number/hex-to-ascii.html) i get this:

```
¹êuêÍ\2MSZ314X

1-3:0.2.8(50)
0-0:1.0.0(240303225425W)
0-0:96.1.1(373335393939323534333030353738303636)
1-0:1.8.1(004730.806*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.8.0(004730.806*kWh)
1-0:2.8.0(000000.000*kWh)
1-0:3.8.0(000000.516*kvarh)
1-0:4.8.0(001782.029*kvarh)
1-0:1.7.0(01.076*kW)
1-0:2.7.0(00.000*kW)
1-0:3.7.0(00.000*kvar)
1-0:4.7.0(00.483*kvar)
1-0:99.97.0(10)(0-0:96.11.63)(231206102343W)(6)(231206102343W)(5)(231206102343W)(4)(231206102007W)(3)(231206102007W)(2)(231206102007W)(1)(231016072545W)(6)(231016075W)(4.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0(30303030303030303030303030303030)
1-0:32.7.0(236.8*V)
1-0:52.7.0(236.4*V)
1-0:72.7.0(235.5*V)
1-0:31.7.0(000.4*A)
1-0:51.7.0(001.9*A)
1-0:71.7.0(003.2*A)
1-0:21.7.0(00.054*kW)
1-0:41.7.0(00.460*kW)
1-0:61.7.0(00.561*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:23.7.0(00.000*kvar)
1-0:43.7.0(00.000*kvar)
1-0:63.7.0(00.000*kvar)
1-0:24.7.0(00.056*kvar)
1-0:44.7.0(00.102*kvar)
1-0:64.7.0(00.324*kvar)
0-1:24.1.0(004)
0-1:96.1.0(3638393339383737)
0-1:24.2.1(240303224401W)(142399.000*kWh)
!F2C8
```

Which looks a lot like a correct message, so I am pretty sure that your connection and all settings are correct. The problem is the beginning of the message which has a bunch of unexpected data, instead of the "/" character indicating the start of an ascii message. Several of the leading characters are in fact not ascii (they are in the range 128-255).

I am not familiar with Swemet and Shenzhen Star Instruments meters. Are these intended for the swedish market? Is there any documentation relevant to the use of their P1 port?

Beaky2000, 2024-03-04:
For reference, this seems to be the same issue: (no solution though)

[](https://forum.currently.one/t/veab-vaxjo-energi-shenzhen-star-stz351-far-crc-mismatch-pa-varje-meddelande/325)

Beaky2000, 2024-03-04:
I get the feeling that the meter is probably buggy and the start of the message is incorrect, which also causes the checksum to not match. If you can capture another message like the one above (with a different checksum) I might be able to figure out a way to calculate the checksum anyway.

riborg, 2024-03-04:
Yes, that is the same meter and same company. Is it possible to disable the CRC like they write in the forum?

Or should I contact the company and ask about the weird messate format?

riborg, 2024-03-04:
Like so? How do I know if it has a different checksum?

```
ea75eacd5c324d535a333134580d0a0d0a312d333a302e322e38283530290d0a
302d303a312e302e302832343033303432313432353157290d0a302d303a3936
2e312e3128333733333335333933393339333233353334333333303330333533
373338333033363336290d0a312d303a312e382e31283030343734382e393333
2a6b5768290d0a312d303a312e382e32283030303030302e3030302a6b576829
0d0a312d303a322e382e31283030303030302e3030302a6b5768290d0a312d30
3a322e382e32283030303030302e3030302a6b5768290d0a302d303a39362e31
342e302830303031290d0a312d303a312e382e30283030343734382e3933332a
6b5768290d0a312d303a322e382e30283030303030302e3030302a6b5768290d
0a312d303a332e382e30283030303030302e3531382a6b76617268290d0a312d
303a342e382e30283030313739322e3631312a6b76617268290d0a312d303a31
2e372e302830312e3538352a6b57290d0a312d303a322e372e302830302e3030
302a6b57290d0a312d303a332e372e302830302e3030302a6b766172290d0a31
2d303a342e372e302830302e3539332a6b766172290d0a312d303a39392e3937
2e302831302928302d303a39362e31312e363329283233313230363130323334
3357292836292832333132303631303233343357292835292832333132303631
3032333433572928342928323331323036313032303037572928332928323331
3230363130323030375729283229283233313230363130323030375729283129
2832333130313630373235343557292836292832333130313630373235343557
3557292834292832333130313630363539343557292833290d0a312d303a3332
2e33322e30283030303030290d0a312d303a35322e33322e3028303030303029
0d0a312d303a37322e33322e30283030303030290d0a312d303a33322e33362e
30283030303030290d0a312d303a35322e33362e30283030303030290d0a312d
303a37322e33362e30283030303030290d0a302d303a39362e31332e30283330
333033303330333033303330333033303330333033303330333033303330290d
0a312d303a33322e372e30283233362e332a56290d0a312d303a35322e372e30
283233352e352a56290d0a312d303a37322e372e30283233332e322a56290d0a
312d303a33312e372e30283030302e342a41290d0a312d303a35312e372e3028
3030322e312a41290d0a312d303a37312e372e30283030352e322a41290d0a31
2d303a32312e372e302830302e3035372a6b57290d0a312d303a34312e372e30
2830302e3530332a6b57290d0a312d303a36312e372e302830312e3032352a6b
57290d0a312d303a32322e372e302830302e3030302a6b57290d0a312d303a34
322e372e302830302e3030302a6b57290d0a312d303a36322e372e302830302e
3030302a6b57290d0a312d303a32332e372e302830302e3030302a6b76617229
0d0a312d303a34332e372e302830302e3030302a6b766172290d0a312d303a36
332e372e302830302e3030302a6b766172290d0a312d303a32342e372e302830
302e3035362a6b766172290d0a312d303a34342e372e302830302e3131352a6b
766172290d0a312d303a36342e372e302830302e3432312a6b766172290d0a30
2d313a32342e312e3028303034290d0a302d313a39362e312e30283336333833
3933333339333833373337290d0a302d313a32342e322e312832343033303432
31323930315729283134323437372e3030302a6b5768290d0a21364630320d0a
```

```
ea75eacd5c324d535a333134580d0a0d0a312d333a302e322e38283530290d0a
302d303a312e302e302832343033303432313433313957290d0a302d303a3936
2e312e3128333733333335333933393339333233353334333333303330333533
373338333033363336290d0a312d303a312e382e31283030343734382e393436
2a6b5768290d0a312d303a312e382e32283030303030302e3030302a6b576829
0d0a312d303a322e382e31283030303030302e3030302a6b5768290d0a312d30
3a322e382e32283030303030302e3030302a6b5768290d0a302d303a39362e31
342e302830303031290d0a312d303a312e382e30283030343734382e3934362a
6b5768290d0a312d303a322e382e30283030303030302e3030302a6b5768290d
0a312d303a332e382e30283030303030302e3531382a6b76617268290d0a312d
303a342e382e30283030313739322e3631362a6b76617268290d0a312d303a31
2e372e302830312e3636362a6b57290d0a312d303a322e372e302830302e3030
302a6b57290d0a312d303a332e372e302830302e3030302a6b766172290d0a31
2d303a342e372e302830302e3630372a6b766172290d0a312d303a39392e3937
2e302831302928302d303a39362e31312e363329283233313230363130323334
3357292836292832333132303631303233343357292835292832333132303631
3032333433572928342928323331323036313032303037572928332928323331
3230363130323030375729283229283233313230363130323030375729283129
2832333130313630373235343557292836292832333130313630372928353557
393435572928332e33322e30283030303030290d0a312d303a35322e33322e30
283030303030290d0a312d303a37322e33322e30283030303030290d0a312d30
3a33322e33362e30283030303030290d0a312d303a35322e33362e3028303030
3030290d0a312d303a37322e33362e30283030303030290d0a302d303a39362e
31332e3028333033303330333033303330333033303330333033303330333033
3033303330290d0a312d303a33322e372e30283233362e342a56290d0a312d30
3a35322e372e30283233352e372a56290d0a312d303a37322e372e3028323333
2e322a56290d0a312d303a33312e372e30283030302e342a41290d0a312d303a
35312e372e30283030322e312a41290d0a312d303a37312e372e30283030352e
352a41290d0a312d303a32312e372e302830302e3035372a6b57290d0a312d30
3a34312e372e302830302e3530342a6b57290d0a312d303a36312e372e302830
312e3130352a6b57290d0a312d303a32322e372e302830302e3030302a6b5729
0d0a312d303a34322e372e302830302e3030302a6b57290d0a312d303a36322e
372e302830302e3030302a6b57290d0a312d303a32332e372e302830302e3030
302a6b766172290d0a312d303a34332e372e302830302e3030302a6b76617229
0d0a312d303a36332e372e302830302e3030302a6b766172290d0a312d303a32
342e372e302830302e3035362a6b766172290d0a312d303a34342e372e302830
302e3131342a6b766172290d0a312d303a36342e372e302830302e3433352a6b
766172290d0a302d313a32342e312e3028303034290d0a302d313a39362e312e
302833363338333933333339333833373337290d0a302d313a32342e322e3128
3234303330343231323930315729283134323437372e3030302a6b5768290d0a
21443746330d0a31323930315729283134323437372e3030302a6b5768290d0a
```

Beaky2000, 2024-03-05:
In the file `components/p1_mini/p1_mini.cpp`

Find the lines:

```
                    if (read_byte == '/') {
                        ESP_LOGD(TAG, "ASCII data format");
```
and change them into:
```
                    if (read_byte == 0xb9) {
                        ESP_LOGD(TAG, "Buggy data format");
```

This should make it accept the messages, but the checksum calculations would fail.

So also find this line:
```
                if (crc == crc_from_msg) {
```

and replace it with:
```
                if (true) {
```

I think this should work fairly well, but I haven't tested it, so let me know how it goes!

riborg, 2024-03-05:
Yay! It worked.. Thank you.. Who need CRC anyway, it's only for cowards.. ;)

[image]

closed 2024-03-05 by Beaky2000