Beaky2000/esphome-p1mini #28: Unknown data format (0xBF). Resetting.

tonzet, 2024-03-13:
Hi, 
I'm getting faultmessage "Unknown data format (0xBF). Resetting."
I've wired my D1 Mini Pro like:
1 - 5V - 5V
2 - RTS - D2
3 - DGND - G
5 - TX - RX
6 - GND - G

Code is according to latest main branch.
RJ12 cable is about 40 cm long.
Meter hardware / supplier is Sagemcom T211 / Ellevio.

Do you see something obviously wrong?

Here are my output from DEBUG:

INFO ESPHome 2024.2.1
INFO Reading configuration /config/p1mini.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing p1reader-esp (board: d1_mini_pro; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5

.....

[14:02:24][I][app:102]: ESPHome version 2024.2.1 compiled on Mar 13 2024, 14:01:51
[14:02:24][C][wifi:577]: WiFi:
[14:02:24][C][wifi:409]:   Local MAC: DC:4F:XX:XX:XX:XX
[14:02:24][C][wifi:414]:   SSID: [redacted]
[14:02:24][C][wifi:415]:   IP Address: 192.168.X.X
[14:02:24][C][wifi:416]:   BSSID: [redacted]
[14:02:24][C][wifi:418]:   Hostname: 'p1reader-esp'
[14:02:24][C][wifi:420]:   Signal strength: -74 dB ▂▄▆█
[14:02:24][C][wifi:424]:   Channel: 6
[14:02:24][C][wifi:425]:   Subnet: 255.255.255.0
[14:02:24][C][wifi:426]:   Gateway: 192.168.X.X
[14:02:24][C][wifi:427]:   DNS1: 192.168.X.X
[14:02:24][C][wifi:428]:   DNS2: 0.0.0.0
[14:02:24][C][logger:447]: Logger:
[14:02:24][C][logger:448]:   Level: DEBUG
[14:02:24][C][logger:449]:   Log Baud Rate: 115200
[14:02:24][C][logger:451]:   Hardware UART: UART0
[14:02:24][C][uart.arduino_esp8266:118]: UART Bus:
[14:02:24][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[14:02:24][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[14:02:24][C][uart.arduino_esp8266:122]:   RX Buffer Size: 512
[14:02:24][C][uart.arduino_esp8266:124]:   Baud Rate: 115200 baud
[14:02:24][C][uart.arduino_esp8266:125]:   Data Bits: 8
[14:02:24][C][uart.arduino_esp8266:126]:   Parity: NONE
[14:02:24][C][uart.arduino_esp8266:127]:   Stop bits: 1
[14:02:24][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[14:02:24][W][uart.arduino_esp8266:143]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
[14:02:24][C][switch.gpio:068]: GPIO Switch 'p1_rts'
[14:02:24][C][switch.gpio:090]:   Restore Mode: always OFF
[14:02:24][C][switch.gpio:031]:   Pin: GPIO4
[14:02:24][C][switch.gpio:068]: GPIO Switch 'status_led'
[14:02:24][C][switch.gpio:090]:   Restore Mode: always OFF
[14:02:24][C][switch.gpio:031]:   Pin: GPIO2
[14:02:24][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'secondary_p1_rts'
[14:02:24][C][gpio.binary_sensor:016]:   Pin: GPIO16
[14:02:24][C][P1Mini:456]: P1 Mini component
[14:02:25][C][captive_portal:088]: Captive Portal:
[14:02:25][C][web_server:168]: Web Server:
[14:02:25][C][web_server:169]:   Address: p1reader-esp:80
[14:02:25][C][mdns:115]: mDNS:
[14:02:25][C][mdns:116]:   Hostname: p1reader-esp
[14:02:25][C][ota:096]: Over-The-Air Updates:
[14:02:25][C][ota:097]:   Address: p1reader-esp:8266
[14:02:25][C][ota:100]:   Using Password.
[14:02:25][C][ota:103]:   OTA version: 2.
[14:02:25][C][api:139]: API Server:
[14:02:25][C][api:140]:   Address: p1reader-esp:6053
[14:02:25][C][api:142]:   Using noise encryption: YES
[14:02:25][C][wifi_signal.sensor:009]: WiFi Signal 'p1reader-esp WiFi Signal'
[14:02:25][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[14:02:25][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[14:02:25][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[14:02:25][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[14:02:25][D][switch:012]: 'p1_rts' Turning ON.
[14:02:25][D][switch:055]: 'p1_rts': Sending state ON
[14:02:25][D][switch:012]: 'status_led' Turning ON.
[14:02:25][D][switch:055]: 'status_led': Sending state ON
[14:02:25][W][P1Mini:139]: Unknown data format (0xBF). Resetting.
[14:02:25][D][switch:016]: 'p1_rts' Turning OFF.
[14:02:25][D][switch:055]: 'p1_rts': Sending state OFF
[14:02:25][D][switch:016]: 'status_led' Turning OFF.
[14:02:25][D][switch:055]: 'status_led': Sending state OFF
[14:02:25][W][P1Mini:448]: Discarding: e5ececbffebabdbbbdbabebcbabfffe18f8e8f8eb8bfb8beb9beb8beb8bcbabe
[14:02:25][W][P1Mini:448]: Discarding: b8bbb9bbb9beb8bababaffbd8f8eb9bfb8beb9bebcbeb8bcb8b8b8babcbfb8b9
[14:02:25][W][P1Mini:448]: Discarding: beb9bebcbeffbebcbeb8bcb8b8b8b8bebfbebebebfbebcbefffffcbd8f8eb9bf
[14:02:25][W][P1Mini:448]: Discarding: b8b8b8b8bebabbbfbeffffbd8f8fb9bfb8bebabebfbeb8bcb8b8b8b8beb8b8b8
[14:02:25][W][P1Mini:448]: Discarding: beffffbd8f8eb9bdb8bebbbebfbeb8bcb8b8b8b8beb8b8b8befffef1fa298f8f
[14:02:25][W][P1Mini:448]: Discarding: b9bfb8bebebebfbeb8bcb8b8b8b8bebebfbfbefffef1fabd8f8eb9bfb8bebab9
[14:02:25][W][P1Mini:448]: Discarding: bfbfbeb8bcb8b8b8b8beb8b8b8beffffbd8f8eb9bfb8bebeb9bebfbeb8bcb8b8
[14:02:25][W][P1Mini:448]: Discarding: b8b8bebfbebdbeffffbd8f8eb9bfb8bebeb1be37bfb8bcb8b8b8b8beb8b8b8be
[14:02:25][W][P1Mini:448]: Discarding: ffffbd8f8eb9bfb8bebababebfbeb8bcb8b8b8b8bfb9bebabfffffbd8f8eb9bf
[14:02:25][W][P1Mini:448]: Discarding: b8bebebabebfbeb8bcb8b8b8b8beb8b8b8beffffbd8d8eb9bdb8bebebabebfbe
[14:02:25][W][P1Mini:448]: Discarding: b8bcb8b8b8b8beb9bfb8beffffbd8d8eb9bfb8bebabbbebfbeb8bcb8b8b8b8be
[14:02:25][W][P1Mini:448]: Discarding: b8b8b8befffef1fabd8f8eb9bfb8bebebbbebfbeb8bcb8b8b8b8beb8b8b8beff
[14:02:25][W][P1Mini:448]: Discarding: fef1fabd8f8eb9bfb8bebebbbebfbeb8bcb8b8b8b8beb8b8b8befffef1fbbd8f
[14:02:25][W][P1Mini:448]: Discarding: 8eb9bfb8bebbbebfbfbeb8bcb8b8b8b8bebabfbebefffff1fabd8f8eb9bfb8be
[14:02:25][W][P1Mini:448]: Discarding: bebebebfbeb8bcb8b8b8b8beb9b9bebefffef1fabd8f8eb9bfb8bebebebebfbe
[14:02:25][W][P1Mini:448]: Discarding: b8bcb8b8b8b8beb8bcbfbefffef1fabd8f8eb9bfb8bebbbbbebfbeb8bcbabbbe
[14:02:25][W][P1Mini:448]: Discarding: bfbebefebd8f8eb9bfb8bebfbabebfbeb8bcbabbbabebcbefea98f8eb9bfb8be
[14:02:25][W][P1Mini:448]: Discarding: bfbabebfbeb8bcbabbbbbebabefebd8f8eb9bfb8bebbb9bebfbeb8bcb8b8b9be
[14:02:25][W][P1Mini:448]: Discarding: bfbee1bd8f8eb9bfb8bebfb9bebfbeb8bcb8b8babebfbee1bd8f8eb9bfb8bebf
[14:02:26][W][P1Mini:448]: Discarding: b9bebfbeb8bcb8b8b9beb8bee1bd8f8eb1bfbfe7bf8f8ee1bd8f8eb9bfb8bebf
[14:02:26][D][sensor:093]: 'p1reader-esp WiFi Signal': Sending state -66.00000 dBm with 0 decimals of accuracy
[14:02:27][D][switch:012]: 'p1_rts' Turning ON.
[14:02:27][D][switch:055]: 'p1_rts': Sending state ON
[14:02:27][D][switch:012]: 'status_led' Turning ON.
[14:02:27][D][switch:055]: 'status_led': Sending state ON
[14:02:27][W][P1Mini:139]: Unknown data format (0xBF). Resetting.
[14:02:27][D][switch:016]: 'p1_rts' Turning OFF.
[14:02:27][D][switch:055]: 'p1_rts': Sending state OFF
[14:02:27][D][switch:016]: 'status_led' Turning OFF.
[14:02:27][D][switch:055]: 'status_led': Sending state OFF
[14:02:27][W][P1Mini:448]: Discarding: b8beb9bebcbeb8bcb8b8b8babcbfb8b9beb9bebcbefffffcbd8d8........

Beaky2000, 2024-03-13:
According to this ...

```
[14:02:24][W][uart.arduino_esp8266:143]: You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
```

... you are not using the latest main branch code... It will never be possible to combine the component with logging (on the same UART)

Other than that, your issues looks a lot like [issue 15](https://github.com/Beaky2000/esphome-p1mini/issues/15), which unfortunately is unresolved.

skemppinen, 2024-05-28:
Hi,
Im having the same issues with Aidon 7534/7492M (Data format is ASCII).

Most of the time, its "No data received for 60 seconds" or "Unknow data format (ax0a). Resetting.  

Occasionally reading values is successfull, but it is very random and not clear why it suddenly is able to get correct values. Even when reading is successfull, I still get errors _Could not parse value from line '/ADN9 7534'_ + _Could not parse value from line '0-0:1.0.0(240528085420S)'_.

[09:54:19][D][sensor:093]: 'p1mini WiFi Signal': Sending state -54.00000 dBm with 0 decimals of accuracy
[09:54:20][D][P1Mini:131]: ASCII data format
[09:54:20][D][P1Mini:217]: CRC verification OK
[09:54:20][D][P1Mini:266]: **Could not parse value from line '/ADN9 7534'**
[09:54:20][D][P1Mini:266]: **Could not parse value from line '0-0:1.0.0(240528085420S)'**
[09:54:20][D][sensor:093]: 'Cumulative Active Import': Sending state 12197.01855 kWh with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Cumulative Active Export': Sending state 832.15802 kWh with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Cumulative Reactive Import': Sending state 0.16200 kvarh with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Cumulative Reactive Export': Sending state 3965.49390 kvarh with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Import': Sending state 0.00000 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Export': Sending state 1.59700 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Export': Sending state 0.23100 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Import Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Export Phase 1': Sending state 0.71400 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Import Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Export Phase 2': Sending state 0.77300 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Import Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Active Export Phase 3': Sending state 0.10500 kW with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Import Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Export Phase 1': Sending state 0.30100 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Import Phase 2': Sending state 0.00000 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Export Phase 2': Sending state 0.22300 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Import Phase 3': Sending state 0.28600 kvar with 3 decimals of accuracy
[09:54:20][D][sensor:093]: 'Momentary Reactive Export Phase 3': Sending state 0.00000 kvar with 3 decimals of accuracy
[09:54:20][D][P1Mini:372]: Cycle times: Identifying = 9485 ms, Message = 62 ms (7 loops), Processing = 102 ms (4 loops), (Total = 9649 ms). 708 bytes in buffer

Beaky2000, 2024-05-28:
The two "Could not parse value from line" messages are normal. It is just simpler to let the parsing fail on these lines than identify them as valid lines that should be ignored. So when you do get a message it is 100% as expected.

If you have not already done that, I would suggest you try disabling the RTS signal (setting it constantly high): https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md

Various other things you could try if you have not already:

- Shorter cable (unless it is already < 20 cm)
- Different ESP module in case it is faulty
- Connect oscilloscope to data line to see what is going on
- etc

skemppinen, 2024-05-28:
Hi,

I shorteded the cable + disabled RTS and results are much better. There are still some unknow dataformat errors, but only just like 1/10 fails. Maybe to get full 100%, Ill try ferrite core wrapped around cable someday, but already now this is working perfectly for me.

Thanks for the code & support!

aahlborg, 2024-07-01:
I got this error on my custom setup (HW based on esphome-p1reader) when activating UART debug with dummy receiver. When disabling the dummy receiver I got CRC errors and warnings about too long processing time from p1_mini. When disabling UART debug completely I got valid results most of the time. I still get 10-20% CRC errors but it could be due to the SW UART having problems with the high bitrate.

Beaky2000, 2024-07-01:
> .... I still get 10-20% CRC errors but it could be due to the SW UART having problems with the high bitrate.

I am not at all surprised that you get some failures when using software UART. I am more surprised that you get *any* data at this baudrate, but it is great that it works! I have not even tried using software for this!

It is also a good tip to anyone having problems to not enable UART debugging, since there is going to be some overhead which could cause problems.

aahlborg, 2024-07-01:
My mistake, it's using hardware UART after all. I thought it was always using software but when the pins are shared with logger then the hardware is used.

Beaky2000, 2025-01-15:
I would recommend anyone having this problem to add a resisitor in the range 1kΩ - 2kΩ between 3.3v and RX on the ESP module, as discussed in [issue 15](https://github.com/Beaky2000/esphome-p1mini/issues/15), and letting me know if it helped or not.

Beaky2000, 2025-02-22:
I have added a "troubleshooting" section to the documentation with information about adding an external resistor in cases like this. Anyone still having problems after trying that should open a new Issue!

closed 2025-02-22 by Beaky2000