Beaky2000/esphome-p1mini #29: Reading multiple meters with different UARTs

Fervensimber, 2024-04-11:
Hello! I have 2 electricity meters connected to 2 UARTs on my ESP32. I used an earlier version of this plugin when it was using custom component and that worked fine by having 2 custom platforms with different UART ids for each. I would now switch to using p1_mini platform from external component, but is there a way to read 2 different UART ports into 2 sets of sensors. Could this support be added? Thanks!

Beaky2000, 2024-04-11:
What happens if you create two UARTs, my_uart_1 and my_uart_2 and then create two p1 minis, p1_mini_1 and p1_mini_2, using the two different UARTs?

I have not tried it, but I would have assumed it would work...

Beaky2000, 2024-04-11:
Ok, I think I have the answer to my own question: There is no way to specify which p1 mini to use for each sensor... Or maybe there is, but I am not aware of it...

Fervensimber, 2024-04-11:
YAML like this is invalid due to duplicate key of p1_mini
```
uart:
  - id: uart_bus_A
    rx_pin:
       number: GPIO03
       inverted: true
    baud_rate: 115200
  - id: uart_bus_B
    rx_pin:
       number: GPIO16
       inverted: true
    baud_rate: 115200
    
p1_mini:
    id: p1_mini_A
    uart_id: uart_bus_A
    minimum_period: 2s       # Should be 0 (zero) if the RTS signal is not used.
    buffer_size: 3072        # Needs to be large enough to hold one entire update from the meter.
    secondary_p1: false
    on_ready_to_receive:
      then:
        - switch.turn_on: status_led
    on_update_received:
      then:
        - switch.turn_off: status_led
    on_communication_error:
      then:
      - switch.turn_off: status_led
p1_mini:
    id: p1_mini_B
    uart_id: uart_bus_B
    minimum_period: 2s       # Should be 0 (zero) if the RTS signal is not used.
    buffer_size: 3072        # Needs to be large enough to hold one entire update from the meter.
    secondary_p1: false
    on_ready_to_receive:
      then:
        - switch.turn_on: status_led
    on_update_received:
      then:
        - switch.turn_off: status_led
    on_communication_error:
      then:
      - switch.turn_off: status_led
 ```
The i2c component does support multiple buses, can you check that component for examples on how to achieve this? https://esphome.io/components/i2c.html

Beaky2000, 2024-04-11:
I will look into it when I have some time to spare!

Beaky2000, 2024-04-15:
I had a look at this during the weekend, but unfortunately I don't understand Python or how it is used by ESPHome. The existing parts I just modified existing files from a different project until they started doing what I wanted and I have no clue how to modify them to allow for more than one p1 mini device.

But if someone who does understand this want to contribute, that would be great and if any modifications are needed for the C++-parts I would be happy to figure that out.

Fervensimber, 2024-04-15:
Thanks for looking into it. If anyone else is having the same issue, my workaround was to read one meter with p1-mini and the other meter using https://esphome.io/components/sensor/dsmr.html DSMR component doesn't have all the data, but it has enough for my use case.

Beaky2000, 2024-06-13:
I believe I have fixed this!

While I am unable to verify with multiple meters, since I only have one, it is now possible to create more than on p1_mini entity using different UARTs!

closed 2024-06-13 by Beaky2000