Beaky2000/esphome-p1mini #32: Allow this component to be used as remote (git) components

bistory, 2024-06-17:
Hi,
Thanks a lot for your great work, I didn't test it yet but I will soon!
I have seen you are suggesting the installation of this component as local component but it should be easier and more practical to use is as remote/git components like suggested in the documentation of Esphome : https://esphome.io/components/external_components

This would allow having one centralized repo for the custom component and one with different yaml configurations depending of the used device (I'm planning to use it with Wemos C3 pico which draw less power the ESP8266 and is more powerful/recent/future proof).

I would also be straightforward to install as it will not require cloning a git repository 👍

Beaky2000, 2024-06-18:
After some quick testing it seems that all that is needed is to add this to a config:

```
external_components:
  - source: github://Beaky2000/esphome-p1mini@main

```

With that the `p1_mini` component and sensor is available.

I am planning to add some documentation about this, but other than that I believe I will keep this project as it is!

bistory, 2024-06-18:
> After some quick testing it seems that all that is needed is to add this to a config:
>
> ```
> external_components:
>   - source: github://Beaky2000/esphome-p1mini@main
> ```
>
> With that the `p1_mini` [...]

Yes, that's how I tested it this morning, installation went flawlessly, I'm also able to use the RGB led of the Wemos C3 Pico (which is cool!) but I made a mistake in my wiring so I have to change it, I will test that tomorrow ;-)

So, once it is done, either I can make a separate repo sharing the yaml file and wiring schemas or we can add the documentation in this repo?
It's up to you 😉

Beaky2000, 2024-06-18:
I have added some basic documentation to point out that this is possible.

I don't think I want to maintain descriptions of other hardware and configurations, so you could add your info as a "discussion" or make your own repo!

closed 2024-06-18 by Beaky2000