Beaky2000/esphome-p1mini #33: Not reading all the data on Sagemcom T211

bistory, 2024-06-19:
Hi,

I have added my ESP32-C3 to the P1 port, the component is reading the data and sending them to Home Assistant, however it seems that all the data are not read.

Here is the relevant log : 

```
[12:48:23][D][switch:012]: 'p1_rts' Turning ON.
[12:48:23][D][switch:055]: 'p1_rts': Sending state ON
[12:48:23][D][light:036]: 'status_led' Setting:
[12:48:23][D][light:059]:   Red: 0%, Green: 100%, Blue: 0%
[12:48:23][D][light:085]:   Transition length: 1.0s
[12:48:23][D][P1Mini:131]: ASCII data format
[12:48:23][D][switch:016]: 'p1_rts' Turning OFF.
[12:48:23][D][switch:055]: 'p1_rts': Sending state OFF
[12:48:23][D][light:036]: 'status_led' Setting:
[12:48:23][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[12:48:23][D][light:085]:   Transition length: 1.0s
[12:48:23][D][P1Mini:217]: CRC verification OK
[12:48:23][D][P1Mini:266]: Could not parse value from line '/FLU5\253972606_B'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.1.4(50221)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.1.1(3153414731323030313933393533)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.1.2()'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:1.0.0(240619124822S)'
[12:48:23][D][P1Mini:273]: No sensor matching: 1.8.1 (0x10801)
[12:48:23][D][P1Mini:273]: No sensor matching: 1.8.2 (0x10802)
[12:48:23][D][P1Mini:273]: No sensor matching: 2.8.1 (0x20801)
[12:48:23][D][P1Mini:273]: No sensor matching: 2.8.2 (0x20802)
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.14.0(0001)'
[12:48:23][D][P1Mini:273]: No sensor matching: 1.4.0 (0x10400)
[12:48:23][D][P1Mini:273]: No sensor matching: 1.6.0 (0x10600)
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:98.1.0(1)(1-0:1.6.0)(1-0:1.6.0)(240301000000W)(240201000000W)(00.000*kW)'
[12:48:23][D][sensor:094]: 'Momentary Active Import': Sending state 0.00000 kW with 3 decimals of accuracy
[12:48:23][D][sensor:094]: 'Momentary Active Export': Sending state 0.81900 kW with 3 decimals of accuracy
[12:48:23][D][sensor:094]: 'Momentary Active Import Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[12:48:23][D][sensor:094]: 'Momentary Active Export Phase 1': Sending state 0.81900 kW with 3 decimals of accuracy
[12:48:23][D][sensor:094]: 'Voltage Phase 1': Sending state 236.89999 V with 1 decimals of accuracy
[12:48:23][D][sensor:094]: 'Current Phase 1': Sending state 3.63000 A with 1 decimals of accuracy
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.3.10(1)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:17.0.0(99.999*kW)'
[12:48:23][D][P1Mini:273]: No sensor matching: 31.4.0 (0x1f0400)
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-1:96.3.10(0)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-2:96.3.10(0)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-3:96.3.10(0)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-4:96.3.10(0)'
[12:48:23][D][P1Mini:266]: Could not parse value from line '0-0:96.13.0()'
[12:48:23][D][P1Mini:380]: Cycle times: Identifying = 44 ms, Message = 70 ms (6 loops), Processing = 71 ms (3 loops), (Total = 185 ms). 666 bytes in buffer
```

I was not able to get a full export of the raw data using the logger in debug mode, so I'm not able to use decode hex blobs at the moment.

The counter is from Belgium, a Sagemcom T211 with S1 port (only one T1), maybe this is a new variation of the counter ?
Thanks for your help !

bistory, 2024-06-19:
Here is an export of the raw data : 
```
2F:46:4C:55:35:5C:32:35:33:39:37:32:36:30:36:5F:42:0D:0A:0D:0A:30:2D:30:3A:39:36:2E:31:2E:34:28:35:30:32:32:31:29:0D:0A:30:2D:30:3A:39:36:2E:31:2E:31:28:33:31:35:33:34:31:34:37:33:31:33:32:33:30:33:30:33:31:33:39:33:33:33:39:33:35:33:33:29:0D:0A:30:2D:30:3A:39:36:2E:31:2E:32:28:29:0D:0A:30:2D:30:3A:31:2E:30:2E:30:28:32:34:30:36:31:39:31:33:31:30:30:37:53:29:0D:0A:31:2D:30:3A:31:2E:38:2E:31:28:30:30:30:30:31:37:2E:33:39:36:2A:6B:57:68:29:0D:0A:31
2D:30:3A:31:2E:38:2E:32:28:30:30:30:30:32:35:2E:34:31:34:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:31:28:30:30:30:30:37:39:2E:33:36:34:2A:6B:57:68:29:0D:0A:31:2D:30:3A:32:2E:38:2E:32:28:30:30:30:30:33:33:2E:37:38:38:2A:6B:57:68:29:0D:0A:30:2D:30:3A:39:36:2E:31:34:2E:30:28:30:30:30:31:29:0D:0A:31:2D:30:3A:31:2E:34:2E:30:28:30:30:2E:30:30:30:2A:6B:57:29:0D:0A:31:2D:30:3A:31:2E:36:2E:30:28:32:34:30:36:31:32:32:32:31:35:30:30:53:29:28:30:32:2E:31
33:33:2A:6B:57:29:0D:0A:30:2D:30:3A:39:38:2E:31:2E:30:28:31:29:28:31:2D:30:3A:31:2E:36:2E:30:29:28:31:2D:30:3A:31:2E:36:2E:30:29:28:32:34:30:33:30:31:30:30:30:30:30:30:57:29:28:32:34:30:32:30:31:30:30:30:30:30:30:57:29:28:30:30:2E:30:30:30:2A:6B:57:29:0D:0A:31:2D:30:3A:31:2E:37:2E:30:28:30:30:2E:30:30:30:2A:6B:57:29:0D:0A:31:2D:30:3A:32:2E:37:2E:30:28:30:30:2E:36:37:37:2A:6B:57:29:0D:0A:31:2D:30:3A:32:31:2E:37:2E:30:28:30:30:2E:30:30:30:2A:6B:57
29:0D:0A:31:2D:30:3A:32:32:2E:37:2E:30:28:30:30:2E:36:37:37:2A:6B:57:29:0D:0A:31:2D:30:3A:33:32:2E:37:2E:30:28:32:33:37:2E:32:2A:56:29:0D:0A:31:2D:30:3A:33:31:2E:37:2E:30:28:30:30:32:2E:39:38:2A:41:29:0D:0A:30:2D:30:3A:39:36:2E:33:2E:31:30:28:31:29:0D:0A:30:2D:30:3A:31:37:2E:30:2E:30:28:39:39:2E:39:39:39:2A:6B:57:29:0D:0A:31:2D:30:3A:33:31:2E:34:2E:30:28:39:39:39:2E:39:39:2A:41:29:0D:0A:30:2D:31:3A:39:36:2E:33:2E:31:30:28:30:29:0D:0A:30:2D:32:3A
```

Decoded using : https://www.rapidtables.com/convert/number/hex-to-ascii.html : 

```
/FLU5\253972606_B

0-0:96.1.4(50221)
0-0:96.1.1(3153414731323030313933393533)
0-0:96.1.2()
0-0:1.0.0(240619131007S)
1-0:1.8.1(000017.396*kWh)
1-0:1.8.2(000025.414*kWh)
1-0:2.8.1(000079.364*kWh)
1-0:2.8.2(000033.788*kWh)
0-0:96.14.0(0001)
1-0:1.4.0(00.000*kW)
1-0:1.6.0(240612221500S)(02.133*kW)
0-0:98.1.0(1)(1-0:1.6.0)(1-0:1.6.0)(240301000000W)(240201000000W)(00.000*kW)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(00.677*kW)
1-0:21.7.0(00.000*kW)
1-0:22.7.0(00.677*kW)
1-0:32.7.0(237.2*V)
1-0:31.7.0(002.98*A)
0-0:96.3.10(1)
0-0:17.0.0(99.999*kW)
1-0:31.4.0(999.99*A)
0-1:96.3.10(0)
0-2:
```

Beaky2000, 2024-06-19:
The Belgian meter obviously outputs different data than the Swedish ones, but you can adjust for this in the config file, to some extent.

Only lines beginning with "1-0:" will be considered and all others will be ignored with a "Could not parse value from line" message. These messages are present with the Swedish meters too for some irrelevant lines in the output.

For the lines beginning with "1-0:" you just have to adjust the sensors in you config. Remove sensors for data that the meter does not output and add new sensors, or adjust the obis code for existing sensors when that makes sense.

bistory, 2024-06-19:
> The Belgian meter obviously outputs different data than the Swedish ones, but you can adjust for this in the config file, to some extent.
>
> Only lines beginning with "1-0:" will be considered and all [...]

Thanks, I finished this configuration, so I put it here for future references. For 1.8.0 and 2.8.0, I had to use combination platform in esphome : 

```
sensor:
- platform: wifi_signal
  name: "${device_name} WiFi Signal"
  update_interval: 10s
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "1.8.1"
  name: "Cumulative Active Import (Peak hours)"
  unit_of_measurement: kWh
  accuracy_decimals: 3
  state_class: "total_increasing"
  device_class: "energy"
  id: p1_181
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "1.8.2"
  name: "Cumulative Active Import (Off-peak hours)"
  unit_of_measurement: kWh
  accuracy_decimals: 3
  state_class: "total_increasing"
  device_class: "energy"
  id: p1_182
- platform: combination
  type: sum
  name: "Cumulative Active Import"
  sources:
    - source: p1_181
    - source: p1_182
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "2.8.1"
  name: "Cumulative Active Export (Peak hours)"
  unit_of_measurement: kWh
  accuracy_decimals: 3
  state_class: "total_increasing"
  device_class: "energy"
  id: p1_281
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "2.8.2"
  name: "Cumulative Active Export (Off-peak hours)"
  unit_of_measurement: kWh
  accuracy_decimals: 3
  state_class: "total_increasing"
  device_class: "energy"
  id: p1_282
- platform: combination
  type: sum
  name: "Cumulative Active Export"
  sources:
    - source: p1_281
    - source: p1_282
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "3.8.0"
#  name: "Cumulative Reactive Import"
#  unit_of_measurement: kvarh
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "4.8.0"
#  name: "Cumulative Reactive Export"
#  unit_of_measurement: kvarh
#  accuracy_decimals: 3
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "1.7.0"
  name: "Momentary Active Import"
  unit_of_measurement: kW
  accuracy_decimals: 3
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "2.7.0"
  name: "Momentary Active Export"
  unit_of_measurement: kW
  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "3.7.0"
#  name: "Momentary Reactive Import"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "4.7.0"
#  name: "Momentary Reactive Export"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "21.7.0"
  name: "Momentary Active Import Phase 1"
  unit_of_measurement: kW
  accuracy_decimals: 3
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "22.7.0"
  name: "Momentary Active Export Phase 1"
  unit_of_measurement: kW
  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "41.7.0"
#  name: "Momentary Active Import Phase 2"
#  unit_of_measurement: kW
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "42.7.0"
#  name: "Momentary Active Export Phase 2"
#  unit_of_measurement: kW
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "61.7.0"
#  name: "Momentary Active Import Phase 3"
#  unit_of_measurement: kW
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "62.7.0"
#  name: "Momentary Active Export Phase 3"
#  unit_of_measurement: kW
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "23.7.0"
#  name: "Momentary Reactive Import Phase 1"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "24.7.0"
#  name: "Momentary Reactive Export Phase 1"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "43.7.0"
#  name: "Momentary Reactive Import Phase 2"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "44.7.0"
#  name: "Momentary Reactive Export Phase 2"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "63.7.0"
#  name: "Momentary Reactive Import Phase 3"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "64.7.0"
#  name: "Momentary Reactive Export Phase 3"
#  unit_of_measurement: kvar
#  accuracy_decimals: 3
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "32.7.0"
  name: "Voltage Phase 1"
  unit_of_measurement: V
  accuracy_decimals: 1
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "52.7.0"
#  name: "Voltage Phase 2"
#  unit_of_measurement: V
#  accuracy_decimals: 1
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "72.7.0"
#  name: "Voltage Phase 3"
#  unit_of_measurement: V
#  accuracy_decimals: 1
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: "31.7.0"
  name: "Current Phase 1"
  unit_of_measurement: A
  accuracy_decimals: 1
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "51.7.0"
#  name: "Current Phase 2"
#  unit_of_measurement: A
#  accuracy_decimals: 1
#- platform: p1_mini
#  p1_mini_id: p1_mini_1
#  obis_code: "71.7.0"
#  name: "Current Phase 3"
#  unit_of_measurement: A
#  accuracy_decimals: 1
#- platform: p1_mini
#  p1_mini_id: p1_mini_2
#  obis_code: "1.8.0"
#  name: "Cumulative Active Import"
#  unit_of_measurement: kWh
#  accuracy_decimals: 3
#  state_class: "total_increasing"
#  device_class: "energy"
#  ...
```

bistory, 2024-06-19:
> Great that is worked out!
> 
> I am not sure, but it might be a good idea to add
> 
> ```
> state_class: "total_increasing"
> device_class: "energy"
> ```
> 
> To your combined sensor...

Good catch, thanks ;-)

bistory, 2024-06-28:
For future reference, here is the repo I created for the ESP32-C3 : https://github.com/tomsbasement/esp32-pico-p1?tab=readme-ov-file
I will probably port it to ESP32-C6 (for WiFi 6)

Beaky2000, 2025-05-24:
> Any way to get both values (date and power) in 1.6.0 ?
> 
> > 1-0:1.6.0(240612221500S)(02.133*kW)
> 
> or eventually only the power value.

At the moment, no. So far, all lines starting with "1-0:" that I have encountered have always contained a single numeric value, so I have assumed this is always the case and as a result, it is not possible to match that line with a text sensor.

It might be a good idea to test this kind of line against the text sensors, if there is no mathing regular sensor. I will have a look at the code and see what can be done.

Anyway, the best you can hope for is getting the entire line as text into HA so you would have to get the relevant information from there yourself.