Beaky2000/esphome-p1mini #35: E360 problems

Fywqdk, 2024-07-03:
Hi! And thanks for sharing this solution!

 I have a Landis+Gyr 360 as well and the same problem as in [issue 22](https://github.com/Beaky2000/esphome-p1mini/issues/22). First off I was unable to run my board (Lolin NodeMCU V3) on the port alone so connected an old USB phone charger. Then I get connection by WIFI and can do OTA updates etc. but there's no data. Tried pulling the RTS to 5V using the NodeMCH boards VUSB but that didn't do anything. I even tried the LED and resistor suggestion from the older issue listed in the previous link. Didn't get anything from 5V to TX but from TX to GND I can light a regular blue LED (with 220 ohm resistor). but nothing at all in terms of data, and if I disconnect the power from the USB side it just dies and cant power on. I am starting to suspect my power company hasn't switched on the port as promised, but I do seem to get at least some power from the various channels.

I am not that strong in this area, so any pointers or ideas would be welcome.

Beaky2000, 2024-07-04:
The fact that you can light an led between TX and ground is not a good sign. TX should be "open drain", meaning that it never outputs any voltage, it only pulls the signal low when active.

I would suggest you try a couple of things directly on the P1-port without the ESP module (unless you already did that):

Led+resistor from 5V to TX (anode on 5V)
Another led+resistor from TX to ground (anode on TX)

Are any of the led:s on or blinking?

connect 5V to RTS. Are there any changes in led behaviour?

Fywqdk, 2024-07-04:
Thanks for the pointers, and sorry I should have been more clear. I did test without the ESP yesterday, but I have a bit more test data now.

LED+resistor only. Only one connection at a time for now.

5V-GND: ON
5V to TX: OFF
TX to ground: ON

5V to RTS: OFF
RTS to ground: ON 

The last two confused me me based on what I read about having to wire RTS to 5V with the E360. I would have expected the opposite results. I will try to breadboard it up to get more reliable testing setup than just me holding the components, and then check more variations including the behaviour of TX when 5V and RTS are connected (or RTS and GND I guess....)

Beaky2000, 2024-07-04:
Yes, lots of confusing readings here... If you don't have a multimeter, get a cheapo one and get some voltages for RTS and TX relative to ground. Even very cheap multimeters are quite useful and good to have around!

Fywqdk, 2024-07-04:
Well this is awkward but it seems I have somehow wired up things wrongly on the RJ12 and that might explain it. My mistake was using a previously soldered resistor + LED that I had laying around and then not taking care to check the direction of the LED so when I first hooked it up to 5V and GND and it lit up I figured I had it worked out right. Turns out I hadn't, it just looked that way due to me misreading the RJ12 pinout orientation (can you tell I am a electronics hobbyist-noob? :D) Anyway I started to suspect something was wrong since pretty much all results were opposite what I expected on my more thorough testing with the breadboard and eventually i tried with a new LED and resistor and suddenly 5V-GND was off but GND-5V was on. Anyway I have to fix the cabling then since currently it's messed up. I guess my problem was cutting an RJ12 cable and then using the wrong end so red became ground on my wires.... ANYWAY..... Will revert back when I hopefully have everything working then. Thanks so far for feedback. It got my brain spinning a bit faster than it apparently did last night.

Fywqdk, 2024-07-04:
Ok that did it. I rewired the RJ12 (5V to data GND - Woops) and plugged it into the ESP with USB power backup to supply 5v to RTS via the VUSB pin on the ESP and then I instantly got results.

Sorry for the inconvenience, but thanks a million again for taking the time to ask questions.

Now I am really happy I didn't write an email yet to the power company asking what the hell was going on...

closed 2024-07-04 by Fywqdk