Beaky2000/esphome-p1mini #36: No data recieved from Landys/Gyr, discarding in logs

rickardfransson, 2024-08-03:
Hi!

I´m getting discarded messages in log:

07:22:39	[W]	[P1Mini:448]	
Discarding: 2f4c474635453336300d0a0d0a302d303a312e302e3028323430383033303632
07:22:39	[W]	[P1Mini:448]	
Discarding: 32333957290d0a312d303a312e382e302830303034303033302e3130342a6b57
07:22:39	[W]	[P1Mini:448]	
Discarding: 68290d0a312d303a322e382e302830303030303030302e3030302a6b5768290d
07:22:39	[W]	[P1Mini:448]	
Discarding: 0a312d303a332e382e302830303030313036392e3039322a6b56417268290d0a
07:22:39	[W]	[P1Mini:448]	
Discarding: 312d303a342e382e302830303030323532342e3438342a6b56417268290d0a31
07:22:39	[W]	[P1Mini:448]	
Discarding: 2d303a312e372e3028303030332e3934372a6b57290d0a312d303a322e372e30
07:22:39	[W]	[P1Mini:448]	
Discarding: 28303030302e3030302a6b57290d0a312d303a332e372e3028303030302e3030
07:22:39	[W]	[P1Mini:448]	
Discarding: 302a6b564172290d0a312d303a342e372e3028303030302e3131352a6b564172
07:22:39	[W]	[P1Mini:448]	
Discarding: 290d0a312d303a32312e372e3028303030332e3930332a6b57290d0a312d303a
07:22:39	[W]	[P1Mini:448]	
Discarding: 32322e372e3028303030302e3030302a6b57290d0a312d303a34312e372e3028
07:22:39	[W]	[P1Mini:448]	
Discarding: 303030302e3033322a6b57290d0a312d303a34322e372e3028303030302e3030
07:22:40	[W]	[P1Mini:448]	
Discarding: 302a6b57290d0a312d303a36312e372e3028303030302e3031322a6b57290d0a
07:22:40	[W]	[P1Mini:448]	
Discarding: 312d303a36322e372e3028303030302e3030302a6b57290d0a312d303a32332e
07:22:40	[W]	[P1Mini:448]	
Discarding: 372e3028303030302e3030302a6b564172290d0a312d303a32342e372e302830
07:22:40	[W]	[P1Mini:448]	
Discarding: 3030302e3031302a6b564172290d0a312d303a34332e372e3028303030302e30
07:22:40	[W]	[P1Mini:448]	
Discarding: 30302a6b564172290d0a312d303a34342e372e3028303030302e3039392a6b56
07:22:40	[W]	[P1Mini:448]	
Discarding: 4172290d0a312d303a36332e372e3028303030302e3030302a6b564172290d0a
07:22:40	[W]	[P1Mini:448]	
Discarding: 312d303a36342e372e3028303030302e3030352a6b564172290d0a312d303a33
07:22:40	[W]	[P1Mini:448]	
Discarding: 322e372e30283233302e392a56290d0a312d303a35322e372e30283233382e30
07:22:40	[W]	[P1Mini:448]	
Discarding: 2a56290d0a312d303a37322e372e30283233352e392a56290d0a312d303a3331
07:22:40	[W]	[P1Mini:448]	
Discarding: 2e372e30283031362e392a41290d0a312d303a35312e372e30283030302e342a
07:22:40	[W]	[P1Mini:448]	
Discarding: 41290d0a312d303a37312e372e30283030302e302a41290d0a21433430300d0a

I´m not getting any info if im visiting the ESP, it only displays NA besides every row.
Also, HA isnt getting any info either.

Found a previous issue where i could turn off CRC, so I edited the p1_mini file and installed by visiting .yaml and clicking install.
But that didnt do the trick.

Full debug log:
<html>
<body>

07:32:05 | [D] | [switch:055] | 'p1_rts': Sending state OFF
07:32:05 | [D] | [switch:016] | 'status_led' Turning OFF.
07:32:05 | [D] | [switch:055] | 'status_led': Sending state OFF
07:32:05 | [D] | [switch:012] | 'p1_rts' Turning ON.
07:32:05 | [D] | [switch:055] | 'p1_rts': Sending state ON
07:32:05 | [D] | [switch:012] | 'status_led' Turning ON.
07:32:05 | [D] | [switch:055] | 'status_led': Sending state ON
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -52.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -50.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -50.00000 dBm with 0 decimals of accuracy
07:32:05 | [W] | [P1Mini:123] | No data received for 60 seconds.
07:32:05 | [D] | [switch:016] | 'p1_rts' Turning OFF.
07:32:05 | [D] | [switch:055] | 'p1_rts': Sending state OFF
07:32:05 | [D] | [switch:016] | 'status_led' Turning OFF.
07:32:05 | [D] | [switch:055] | 'status_led': Sending state OFF
07:32:05 | [W] | [P1Mini:448] | Discarding: 2f4c474635453336300d0a0d0a302d303a312e302e3028323430383033303633
07:32:05 | [W] | [P1Mini:448] | Discarding: 31343457290d0a312d303a312e382e302830303034303033302e3731352a6b57
07:32:05 | [W] | [P1Mini:448] | Discarding: 68290d0a312d303a322e382e302830303030303030302e3030302a6b5768290d
07:32:05 | [W] | [P1Mini:448] | Discarding: 0a312d303a332e382e302830303030313036392e3039322a6b56417268290d0a
07:32:05 | [W] | [P1Mini:448] | Discarding: 312d303a342e382e302830303030323532342e3439342a6b56417268290d0a31
07:32:05 | [W] | [P1Mini:448] | Discarding: 2d303a312e372e3028303030342e3035332a6b57290d0a312d303a322e372e30
07:32:05 | [W] | [P1Mini:448] | Discarding: 28303030302e3030302a6b57290d0a312d303a332e372e3028303030302e3030
07:32:05 | [W] | [P1Mini:448] | Discarding: 302a6b564172290d0a312d303a342e372e3028303030302e3033372a6b564172
07:32:05 | [W] | [P1Mini:448] | Discarding: 290d0a312d303a32312e372e3028303030342e3030392a6b57290d0a312d303a
07:32:05 | [W] | [P1Mini:448] | Discarding: 32322e372e3028303030302e3030302a6b57290d0a312d303a34312e372e3028
07:32:05 | [W] | [P1Mini:448] | Discarding: 303030302e3033312a6b57290d0a312d303a34322e372e3028303030302e3030
07:32:05 | [W] | [P1Mini:448] | Discarding: 302a6b57290d0a312d303a36312e372e3028303030302e3031322a6b57290d0a
07:32:05 | [W] | [P1Mini:448] | Discarding: 312d303a36322e372e3028303030302e3030302a6b57290d0a312d303a32332e
07:32:05 | [W] | [P1Mini:448] | Discarding: 372e3028303030302e3036372a6b564172290d0a312d303a32342e372e302830
07:32:05 | [W] | [P1Mini:448] | Discarding: 3030302e3030302a6b564172290d0a312d303a34332e372e3028303030302e30
07:32:05 | [W] | [P1Mini:448] | Discarding: 30302a6b564172290d0a312d303a34342e372e3028303030302e3039382a6b56
07:32:05 | [W] | [P1Mini:448] | Discarding: 4172290d0a312d303a36332e372e3028303030302e3030302a6b564172290d0a
07:32:05 | [W] | [P1Mini:448] | Discarding: 312d303a36342e372e3028303030302e3030352a6b564172290d0a312d303a33
07:32:05 | [W] | [P1Mini:448] | Discarding: 322e372e30283233302e382a56290d0a312d303a35322e372e30283233362e38
07:32:05 | [W] | [P1Mini:448] | Discarding: 2a56290d0a312d303a37322e372e30283233352e312a56290d0a312d303a3331
07:32:05 | [W] | [P1Mini:448] | Discarding: 2e372e30283031372e332a41290d0a312d303a35312e372e30283030302e342a
07:32:05 | [W] | [P1Mini:448] | Discarding: 41290d0a312d303a37312e372e30283030302e302a41290d0a21433438360d0a
07:32:05 | [D] | [switch:012] | 'p1_rts' Turning ON.
07:32:05 | [D] | [switch:055] | 'p1_rts': Sending state ON
07:32:05 | [D] | [switch:012] | 'status_led' Turning ON.
07:32:05 | [D] | [switch:055] | 'status_led': Sending state ON
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:05 | [W] | [api.connection:104] | ESPHome Logs 2024.7.3 (192.168.1.231): Connection closed
07:32:05 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -51.00000 dBm with 0 decimals of accuracy
07:32:14 | [D] | [sensor:093] | 'p1mini WiFi Signal': Sending state -50.00000 dBm with 0 decimals of accuracy
07:32:24 | [D] | [sensor:093] |  

<br class="Apple-interchange-newline">
</body>
</html>

Beaky2000, 2024-08-03:
There is a known issue with the Landis & Gyr E360 and the RTS signal. Make sure to read about that meter on [this page](https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md#landisgyr-e360).

rickardfransson, 2024-08-04:
Hi!
So I took the RTS cabel and soldered it to +5V pin on the ESP. And after that i went in to the .yaml and erased - switch.turn_off: p1_rts .
Found that line on 3 places. Maybe I did something wrong here?

I also changed to 0s in minimum period.

But, I get discarded in logs despite all this. And NA in every row when i visit the dongel.

Full logg:

INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/p1mini.yaml...
INFO Starting log output from 192.168.1.245 using esphome API
INFO Successfully connected to p1mini @ 192.168.1.245 in 0.056s
INFO Successful handshake with p1mini @ 192.168.1.245 in 2.455s
[10:51:18][I][app:100]: ESPHome version 2024.7.3 compiled on Aug  4 2024, 10:46:20
[10:51:18][C][wifi:599]: WiFi:
[10:51:18][C][wifi:427]:   Local MAC: 7C:87:CE:B5:97:52
[10:51:18][C][wifi:432]:   SSID: [redacted]
[10:51:18][C][wifi:435]:   IP Address: 192.168.1.245
[10:51:18][C][wifi:438]:   BSSID: [redacted]
[10:51:18][C][wifi:440]:   Hostname: 'p1mini'
[10:51:18][C][wifi:442]:   Signal strength: -57 dB ▂▄▆█
[10:51:18][C][wifi:446]:   Channel: 1
[10:51:18][C][wifi:447]:   Subnet: 255.255.255.0
[10:51:18][C][wifi:448]:   Gateway: 192.168.1.1
[10:51:18][C][wifi:449]:   DNS1: 192.168.1.1
[10:51:18][C][wifi:450]:   DNS2: 0.0.0.0
[10:51:18][C][logger:185]: Logger:
[10:51:18][C][logger:186]:   Level: DEBUG
[10:51:18][C][logger:188]:   Log Baud Rate: 0
[10:51:18][C][logger:189]:   Hardware UART: UART0
[10:51:18][C][uart.arduino_esp8266:118]: UART Bus:
[10:51:18][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[10:51:18][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[10:51:18][C][uart.arduino_esp8266:122]:   RX Buffer Size: 512
[10:51:18][C][uart.arduino_esp8266:124]:   Baud Rate: 115200 baud
[10:51:18][C][uart.arduino_esp8266:125]:   Data Bits: 8
[10:51:18][C][uart.arduino_esp8266:126]:   Parity: NONE
[10:51:18][C][uart.arduino_esp8266:127]:   Stop bits: 1
[10:51:18][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[10:51:18][C][switch.gpio:068]: GPIO Switch 'p1_rts'
[10:51:18][C][switch.gpio:090]:   Restore Mode: always OFF
[10:51:18][C][switch.gpio:031]:   Pin: GPIO4
[10:51:18][C][switch.gpio:068]: GPIO Switch 'status_led'
[10:51:18][C][switch.gpio:090]:   Restore Mode: always OFF
[10:51:18][C][switch.gpio:031]:   Pin: GPIO2
[10:51:18][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'secondary_p1_rts'
[10:51:18][C][gpio.binary_sensor:016]:   Pin: GPIO16
[10:51:18][C][P1Mini:456]: P1 Mini component
[10:51:18][C][captive_portal:088]: Captive Portal:
[10:51:18][C][web_server:173]: Web Server:
[10:51:18][C][web_server:174]:   Address: p1mini.local:80
[10:51:18][C][mdns:116]: mDNS:
[10:51:18][C][mdns:117]:   Hostname: p1mini
[10:51:18][C][esphome.ota:073]: Over-The-Air updates:
[10:51:18][C][esphome.ota:074]:   Address: p1mini.local:8266
[10:51:18][C][esphome.ota:075]:   Version: 2
[10:51:18][C][esphome.ota:078]:   Password configured
[10:51:18][C][safe_mode:018]: Safe Mode:
[10:51:18][C][safe_mode:019]:   Boot considered successful after 60 seconds
[10:51:18][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:51:18][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[10:51:18][C][api:139]: API Server:
[10:51:18][C][api:140]:   Address: p1mini.local:6053
[10:51:18][C][api:142]:   Using noise encryption: YES
[10:51:18][C][wifi_signal.sensor:009]: WiFi Signal 'p1mini WiFi Signal'
[10:51:18][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[10:51:18][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[10:51:18][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[... 408 lines cut ...]
  p1_mini_id: p1_mini_1
  obis_code: 31.7.0
  name: Current Phase 1
  unit_of_measurement: A
  accuracy_decimals: 1
  disabled_by_default: false
  force_update: false
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: 51.7.0
  name: Current Phase 2
  unit_of_measurement: A
  accuracy_decimals: 1
  disabled_by_default: false
  force_update: false
- platform: p1_mini
  p1_mini_id: p1_mini_1
  obis_code: 71.7.0
  name: Current Phase 3
  unit_of_measurement: A
  accuracy_decimals: 1
  disabled_by_default: false
  force_update: false

INFO Configuration is valid!

rickardfransson, 2024-08-04:
Found the problem. I had reverted CRC, so I took back CRC check and now it works!!

Beaky2000, 2024-08-04:
#

> Found the problem. I had reverted CRC, so I took back CRC check and now it works!!

Ah, good! I was beginning to get really confused about what was happening here!

closed 2024-08-04 by Beaky2000