Beaky2000/esphome-p1mini #37: Secondary device

HenkeGoransson, 2024-08-09:
When connecting a second device according to the schematics and with a 2kohm resistor the d1mini is not reading data anymore.
The second device gets data.
This d1mini still works and reports wifi data.
Error message in the log: "no data received for 60s"

When disconnecting the second device the d1mini works again.
Is this a resistor issue? Higher or lower?

Beaky2000, 2024-08-10:
I don't think the resistor is the problem. In fact, since the RTS from the second device is ignored, you could remove the resistor and the LED and it wouldn't (shouldn't) make any difference.

When you say that disconnecting the second device, it starts working again, do you mean physically disconnecting it or disabling it in the yaml or both?

I can think of two possible issues:

The second device is incorrectly connected. The signal from the meter should go to RX on the D1 mini and the signal to the second device should go from TX on the D1 mini. These signals should not be directly connected!

or

Your meter has some issue with the use of the RTS signal. Read this and see if there is any mention of your meter: https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md . If this is the case, you would be getting a lot of "Discarding ... " messages in the log.

HenkeGoransson, 2024-08-10:
Didn't understand that it was supposed to go from TX on the D1 mini. Now the D1mini still works when connecting the second device. 
However, how does the D1mini know to send data through?
It doesn't seem like the second device gets any data.

Beaky2000, 2024-08-10:
There is a setting in the yaml:

 `    secondary_p1: false`

... that you need to change to `true`. I should probably make that clearer in the documentation!

HenkeGoransson, 2024-08-10:
Yes!
now its working like a charm!
I should have seen that option.

Perfect! Thanks you soo much!
way better than buying a splitter for 40€ ;)

closed 2024-08-10 by HenkeGoransson