Beaky2000/esphome-p1mini #38: Stability issue Sagemcom T211

phlundblom, 2024-09-16:
Ran into some kind of strange stability issue with my T211 meter. After a few hours the ESP32 device (Wemos S2 Mini) hangs and disconnects from Wifi. Only way to resurrect it is by power cycle the microcontroller.

Seems like this happens quicker with a lower polling interval. I use she esphome-p1mini as an external component.

As for reference I removed the external component inclusion + configuration from the same ESP Home Configuration and just added a `wifi_signal` sensor.  That setup has been running for a week now without any problems.

Any ideas before I start dissecting the code and debug?

Beaky2000, 2024-09-16:
Checking if there is anything in the log would be a good start and if not, set the log level to debug and check again.

Once it hangs, I assume you can no longer access the log, but see if there is a way to continously capture the log!

Since it is an ESP32 I would try to power it from a separate source in case there is a problem with the power consumption!

phlundblom, 2024-10-07:
I finally found time to investigate this further. My conclusion is that all my problems were caused by the cheap Lolin S2 Mini clone that I had used. This is not the first one of these where the 3.3V regulator has caused headaches. Since switching to proper ESP32 hardware, it has been 100% stable. Sorry for the noise.

closed 2024-10-07 by phlundblom