Beaky2000/esphome-p1mini #39: Aidon 6534 missing first part of message

mkarls20, 2024-09-21:
I'm too experiencing the same issue. 

I managed to get some data out of it though by forcing the code to read the data as ASCII. Looks like the first lines of the message gets corrupted somehow. 

I have an oscilloscope but not too much experience using one. I will hook it up during the weekend and see what I can get out of it. @Beaky2000 any tips what I should be looking for?

### My setup

- Aidon 6534 (2023) with a 6490 M module
- ESP32 (only esp I had laying around with external antenna that is needed to get a signal out of the metal enclosure where my meeter is located)
- 1k pull-up resistor to 3.3V on the Data pin
- Data ground connected to ESP ground
- RTS connected to GPIO21
- 10 cm cable between HAN-port and ESP32

### Output with original code
```terminal
[21:40:33][C][logger:185]: Logger:
[21:40:33][C][logger:186]:   Level: DEBUG
[21:40:33][C][logger:188]:   Log Baud Rate: 0
[21:40:33][C][logger:189]:   Hardware UART: UART0
[21:40:33][C][uart.arduino_esp32:151]: UART Bus 0:
[21:40:33][C][uart.arduino_esp32:152]:   TX Pin: GPIO1
[21:40:33][C][uart.arduino_esp32:153]:   RX Pin: GPIO3
[21:40:33][C][uart.arduino_esp32:155]:   RX Buffer Size: 512
[21:40:33][C][uart.arduino_esp32:157]:   Baud Rate: 115200 baud
[21:40:33][C][uart.arduino_esp32:158]:   Data Bits: 8
[21:40:33][C][uart.arduino_esp32:159]:   Parity: NONE
[21:40:33][C][uart.arduino_esp32:160]:   Stop bits: 1
[21:40:33][C][switch.gpio:068]: GPIO Switch 'p1_rts'
[21:40:33][C][switch.gpio:091]:   Restore Mode: always OFF
[21:40:33][C][switch.gpio:031]:   Pin: GPIO21
[21:40:33][C][P1Mini:456]: P1 Mini component
[21:40:33][C][captive_portal:088]: Captive Portal:
[21:40:33][C][web_server:145]: Web Server:
[21:40:33][C][web_server:146]:   Address: esph-p1mini.home.arpa:80
[21:40:33][C][mdns:116]: mDNS:
[21:40:33][C][mdns:117]:   Hostname: esph-p1mini
[21:40:33][C][esphome.ota:073]: Over-The-Air updates:
[21:40:33][C][esphome.ota:074]:   Address: esph-p1mini.home.arpa:3232
[21:40:33][C][esphome.ota:075]:   Version: 2
[21:40:33][C][esphome.ota:078]:   Password configured
[21:40:33][C][safe_mode:018]: Safe Mode:
[21:40:33][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:40:33][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:40:33][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:40:33][C][api:139]: API Server:
[21:40:33][C][api:140]:   Address: esph-p1mini.home.arpa:6053
[21:40:33][C][api:142]:   Using noise encryption: YES
[21:40:33][C][wifi_signal.sensor:010]: WiFi Signal 'esph-p1mini WiFi Signal'
[21:40:33][C][wifi_signal.sensor:010]:   Device Class: 'signal_strength'
[21:40:33][C][wifi_signal.sensor:010]:   State Class: 'measurement'
[21:40:33][C][wifi_signal.sensor:010]:   Unit of Measurement: 'dBm'
[21:40:33][C][wifi_signal.sensor:010]:   Accuracy Decimals: 0
[21:40:36][D][sensor:094]: 'esph-p1mini WiFi Signal': Sending state -75.00000 dBm with 0 decimals of accuracy
[21:40:39][W][P1Mini:139]: Unknown data format (0xEF). Resetting.
[21:40:39][D][switch:016]: 'p1_rts' Turning OFF.
[21:40:39][D][switch:055]: 'p1_rts': Sending state OFF
[21:40:39][W][P1Mini:448]: Discarding: effafb5a57c1fb68290d0a312d303a342e382e302830303030303238392e3737
[21:40:39][W][P1Mini:448]: Discarding: 312a6b56417268290d0a312d303a312e372e3028303030322e3731312a6b5729
[21:40:39][W][P1Mini:448]: Discarding: 0d0a312d303a322e372e3028303030302e3030302a6b57290d0a312d303a332e
[21:40:39][W][P1Mini:448]: Discarding: 372e3028303030302e3030302a6b564172290d0a312d303a342e372e30283030
[21:40:39][W][P1Mini:448]: Discarding: 30302e3032302a6b564172290d0a312d303a32312e372e3028303030302e3433
[21:40:39][W][P1Mini:448]: Discarding: 352a6b57290d0a312d303a32322e372e3028303030302e3030302a6b57290d0a
[21:40:39][W][P1Mini:448]: Discarding: 312d303a34312e372e3028303030322e3133382a6b57290d0a312d303a34322e
[21:40:39][W][P1Mini:448]: Discarding: 372e3028303030302e3030302a6b57290d0a312d303a36312e372e3028303030
[21:40:39][W][P1Mini:448]: Discarding: 302e3133302a6b57290d0a312d303a36322e372e3028303030302e3030302a6b
[21:40:39][W][P1Mini:448]: Discarding: 57290d0a312d303a32332e372e3028303030302e3138342a6b564172290d0a31
[21:40:39][W][P1Mini:448]: Discarding: 2d303a32342e372e3028303030302e3030302a6b564172290d0a312d303a3433
[21:40:39][W][P1Mini:448]: Discarding: 2e372e3028303030302e3030302a6b564172290d0a312d303a34342e372e3028
[... 86 lines cut ...]
[22:01:59][D][sensor:094]: 'Momentary Active Export': Sending state 0.00000 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Import': Sending state 0.00000 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Export': Sending state 0.15100 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Import Phase 1': Sending state 0.14300 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Export Phase 1': Sending state 0.00000 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Import Phase 2': Sending state 0.50700 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Export Phase 2': Sending state 0.00000 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Import Phase 3': Sending state 0.05400 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Active Export Phase 3': Sending state 0.00000 kW with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Import Phase 1': Sending state 0.09300 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Export Phase 1': Sending state 0.00000 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Import Phase 2': Sending state 0.00000 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Export Phase 2': Sending state 0.18700 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Import Phase 3': Sending state 0.00000 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Momentary Reactive Export Phase 3': Sending state 0.05300 kvar with 3 decimals of accuracy
[22:01:59][D][sensor:094]: 'Voltage Phase 1': Sending state 234.80000 V with 1 decimals of accuracy
[22:01:59][D][sensor:094]: 'Voltage Phase 2': Sending state 233.20000 V with 1 decimals of accuracy
[22:01:59][D][sensor:094]: 'Voltage Phase 3': Sending state 235.60001 V with 1 decimals of accuracy
[22:01:59][D][sensor:094]: 'Current Phase 1': Sending state 0.70000 A with 1 decimals of accuracy
[22:01:59][D][sensor:094]: 'Current Phase 2': Sending state 2.30000 A with 1 decimals of accuracy
[22:01:59][D][sensor:094]: 'Current Phase 3': Sending state 0.30000 A with 1 decimals of accuracy
[22:01:59][D][P1Mini:380]: Cycle times: Identifying = 9960 ms, Message = 41 ms (3 loops), Processing = 86 ms (3 loops), (Total = 10087 ms). 584 bytes in buffer
```

_Originally posted by @mkarls20 in https://github.com/Beaky2000/esphome-p1mini/issues/15#issuecomment-2362101620_

mkarls20, 2024-09-21:
Additionally I've also tried increasing the `rx_buffer_size` and the `buffer_size` in the yaml without an changes to the outcome.

Beaky2000, 2024-09-21:
I would suggest trying to set RTS always high to see if that makes any difference! 

[The RTS signal](https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md)

mkarls20, 2024-09-21:
I got some time to put the scope on it this afternoon. My finding is that the HIGH signals are not high enough I guess. I played around with different values for the pull-up resistor on the data line. I just kept adding resistors in parallel until I got more data. Still missing the first couple of characters though so the CRC check do not work. 

### 1k&Omega; resistor and forced ASCII mode

This was my original setup and the message that gets thru is the same. This is what the scope looked like. The yellow line is the RTS and the blue is the data line. The data lines first couple of 

#### Scope
[image]

### 150 &Omega; and forced ASCII mode

Better looking data signal and also catching more of the data but not all of it. Looks like most of the time it is just missing the first few characters.

#### Scope
[image]

#### Message

```terminal
21:00:19][D][switch:055]: 'p1_rts': Sending state ON
[21:00:20][D][P1Mini:131]: ASCII data format
[21:00:20][D][switch:016]: 'p1_rts' Turning OFF.
[21:00:20][D][switch:055]: 'p1_rts': Sending state OFF
[21:00:20][W][P1Mini:231]: CRC mismatch, calculated E30F != 1D28. Message ignored.
[21:00:20][D][P1Mini:233]: Buffer:
S\xa2

0-0:1.0.0(240921200020W)
1-0:1.8.0(00002850.302*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000351.620*kVArh)
1-0:4.8.0(00000302.443*kVArh)
1-0:1.7.0(0001.208*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.615*kVAr)
1-0:4.7.0(0000.000*kVAr)
1-0:21.7.0(0000.533*kW)
1-0:22.7.0(0000.000*kW)
1-0:41.7.0(0000.607*kW)
1-0:42.7.0(0000.000*kW)
1-0:61.7.0(0000.062*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.480*kVAr)
1-0:24.7.0(0000.000*kVAr)
1-0:43.7.0(0000.189*kVAr
[21:00:20][D][switch:016]: 'p1_rts' Turning OFF.
```

Should I go even lower on the resistor? Don't want the meeter to have to sink to much current. According to [spec](https://aidon.com/wp-content/uploads/2023/06/AIDONFD_RJ12_HAN_Interface_SV.pdf) it sinks maximum of 30 mA and I'm closing in on the limit. It also says it is protected so I guess there is no harm in trying.

mkarls20, 2024-09-21:
> I would suggest trying to set RTS always high to see if that makes any difference!
> 
> [The RTS signal](https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md)

I'll go try it right away...

mkarls20, 2024-09-21:
> I would suggest trying to set RTS always high to see if that makes any difference!
> 
> [The RTS signal](https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md)

I hard wired the RTS to +5V from the meeter. No change. Still missing the first few characters.

mkarls20, 2024-09-21:
Problem solved. It was the resistor. I put a 120 &Omega; resistor in parallell with my original 1k&Omega; resistor and now it works. 

The 1k value I originally got from the esphome-p1reader project but didn't work on my Olimex ESP32 devkit lipo-board.

closed 2024-09-21 by mkarls20

Beaky2000, 2024-09-22:
@mkarls20, I am glad that you got it to work, but having that low pull-up resistance (120 | 1000 | whatever the ESP uses internally) is pretty crazy!

In case you find this as fascinating as I do, it would be very interesting to see the signal on the scope without the ESP connected at all. So just a (reasonable) pull-up between 5V and TX on the port and the scope between data-ground and TX...