Beaky2000/esphome-p1mini #41: Unknown data format (0xF0)

N1S5E, 2024-12-02:
I am getting the following message when plugging in a Wemos D1-mini (Brand new).

15:01:09 | [W] | [P1Mini:139] | Unknown data format (0xF0). Resetting.

The meter is a KAIFA MA304T4E which is KAIFA MA304H4E but for higher Ampererage.
The cable is like 10cm so should be short enough.

Some more messages if that is of any help.
14:33:40	[W]	[P1Mini:139]	Unknown data format (0xF0). Resetting.
14:33:40	[D]	[switch:016]	'p1_rts' Turning OFF.
14:33:40	[D]	[switch:055]	'p1_rts': Sending state OFF
14:33:40	[D]	[switch:016]	'status_led' Turning OFF.
14:33:40	[D]	[switch:055]	'status_led': Sending state OFF
14:33:42	[D]	[switch:012]	'p1_rts' Turning ON.
14:33:42	[D]	[switch:055]	'p1_rts': Sending state ON
14:33:42	[D]	[switch:012]	'status_led' Turning ON.
14:33:42	[D]	[switch:055]	'status_led': Sending state ON

Beaky2000, 2024-12-02:
So no "Discarding ..." messages in the log? How frequently do you get the "Unknown data format ..."? Is it always 0xF0?

Try setting [RTS always high](https://github.com/Beaky2000/esphome-p1mini/blob/main/NO-RTS.md) and see if it makes any difference!

N1S5E, 2024-12-02:
No Discarding at all, only 0xF0 at a frequency of 2 seconds.
Just goes on and on.

Will try to set RTS Always High tomorrow.

N1S5E, 2024-12-03:
RTS Always High is set and now it works like a charm.

Also added the device to ESPHome in Home Assistant so now I can do OTA-updates directly, took a few days to figure that out 😂

closed 2024-12-03 by N1S5E