Beaky2000/esphome-p1mini #42: Could not parse value from line '0-0:96.14.0(0001)'

rj2s, 2024-12-27:
Hi,
Just tested yesterday, all working fine except in HA (ESPHome).

Except I can't read the the 96.14.0 (tarif code = 1 for Day, 2 for night)
Got the same error for other data:
[13:07:29][D][P1Mini:266]: Could not parse value from line '/FLU5\253770234_A'
[13:07:29][D][P1Mini:266]: Could not parse value from line '0-0:96.1.4(50217)'
[13:07:29][D][P1Mini:266]: Could not parse value from line '0-0:96.1.1(3153414731313030313939343531)'
[13:07:29][D][P1Mini:266]: Could not parse value from line '0-0:1.0.0(241227130754W)'
I guess bescause data are not value but text ?
How to use this ?

Second question is about handling such data:
[13:07:33][D][P1Mini:266]: Could not parse value from line '0-0:98.1.0(13)(1-0:1.6.0)(1-0:1.6.0)(231201000000W)(231119174500W)(05.178*kW)(240101000000W)(231210190000W)(06.057*kW)(240201000000W)(240113191500W)(05.670*kW)(240301000000W)(240204190000W)(05.078*kW)(240401000000S)(240309160000W)(06.621*kW)(240501000000S)(240412230000S)(03.783*kW)(240601000000S)(240525210000S)(04.730*kW)(240701000000S)(240623201500S)(05.022*kW)(240801000000S)(240729210000S)(04.120*kW)(240901000000S)(240811181500S)(05.364*kW)(2410010
98.1.0 is the max current for the last 12 months...

Using 'ESPHome version 2024.12.2 compiled on Dec 27 2024, 12:55:46'

Best regards,
Roland

rj2s, 2024-12-27:
Hi, I'm leaving in Belgium ... it's a meter 'Siconia S211' from ORES.
Goal is to have a the information about the 'Tarif rate' used (day or night) and this value give me the right info.
Can you help me ?

Beaky2000, 2024-12-27:
How does the tarif information look in the message? I don't see it in your examples. There is "0-0:96.1.4", is that what you mean?

rj2s, 2024-12-27:
The info provided is like this (used in an old integration): 
| replace('0001', 'Jour') | replace('0002', 'Nuit')

Beaky2000, 2024-12-27:
Ok, it is currently not possible to get this information, but I am considering adding a string sensor that could be used to match any line and send the result as text.

I am not making any promisses though...

rj2s, 2024-12-27:
Super, I can wait, don't worry :-)

Anyway, many thanks for this super job !

Beaky2000, 2024-12-30:
I have created a new branch "text_sensor" that you can try out: [LINK](https://github.com/Beaky2000/esphome-p1mini/tree/text_sensor)

There you should be able to create a text sensor, like this:

```
text_sensor:
  - platform: p1_mini
    name: "Tarif rate"
    p1_mini_id: p1_mini_1
    identifier: "0-0:96.14.0(000"
```

This will give you the whole line as text. You could probably filter it like this:

```
    filters:
      - map:
        - 0-0:96.14.0(0001) -> Jour
        - 0-0:96.14.0(0002) -> Nuit
```

Try it out and let me know if it works!

rj2s, 2025-01-11:
Hi & Happy new year !

just tested, changing the source to  "github://Beaky2000/esphome-p1mini@text_sensor"  :-)

like this:

```
text_sensor:
  - platform: p1_mini
    name: "ESPHome_P1 Tarif rate"
    p1_mini_id: p1_mini_1
    identifier: "0-0:96.14.0(000"
    filters:
      - map:
        - 0-0:96.14.0(0001) -> Jour
        - 0-0:96.14.0(0002) -> Nuit
  - platform: p1_mini
    name: "ESPHome_P1 Meter ID"
    icon: "mdi:identifier"
    p1_mini_id: p1_mini_1
    identifier: "/"
    filters:
      - substitute: "/ -> "
  - platform: p1_mini
    name: "ESPHome_P1 Date+Time"
    icon: "mdi:calendar-range"
    p1_mini_id: p1_mini_1
    identifier: "0-0:1.0.0("
    filters:
      - substitute: "0-0:1.0.0( -> "
      - substitute: ") -> "
```
    
Working like expected, now got new sensors : 
'ESPHome_P1 Tarif rate' -> 'Nuit' (normal as today saturday 11/Jan is considered as Nuit/Low tarif) 
'ESPHome_P1 Meter ID' -> 'FLU5\253770234'
'ESPHome_P1 Date+Time' ->  '250111113638'

Excellent ! just what I'm looking for, bravo !
Regards

Beaky2000, 2025-01-11:
Great! Since this is working for me too I will merge this to main.

closed 2025-01-11 by Beaky2000