Beaky2000/esphome-p1mini #49: Transition from p1reader - Unknown data format (0x68). Resetting.

zxmon21, 2025-02-21:
Hi,
I have used a ESP32 with attached p1reader-YAML to read from my meter, which has worked fine. This setup can no longer be upgraded due to custom component being deprecated in esphome.
I merged my "old" YAML with the external component, resulting in the attached new YAML. That compiles fine and uploads, but results in Unknown data format (0x68). I'm attaching a log.

[logs_esp32-power-meter-monitor_run.txt](https://github.com/user-attachments/files/18907514/logs_esp32-power-meter-monitor_run.txt)

[ESPhome_configs.zip](https://github.com/user-attachments/files/18907571/ESPhome_configs.zip)

Beaky2000, 2025-02-21:
You should not include code from another project into this one (includes: - p1reader.h)!

Are you inverting the singal in hardware (with a transistor)? If so, you should not invert in the config.

If you are not using the RTS signal, you need to set minimum_period to 0s.

zxmon21, 2025-02-21:
Good catch! It was the inversion/pullup pointed out in the comments... It's working now. Sorry, there's clearly a lot of electronics to learn 👍  I'm attaching the now working YAML.

[p1reader_working.zip](https://github.com/user-attachments/files/18916193/p1reader_working.zip)

closed 2025-02-21 by zxmon21