Beaky2000/esphome-p1mini #63: ESP8266 + E360 gives unknown data format (0xe7)

thedevleon, 2025-06-11:
Hello,

I'm using a WEMOS D1 mini with the p1mini config (with some very slight adjustments) together with a Landis+Gyr E360 smart meter.

I have RTS hooked up to 5V, TX to D7 and DGND to GND. No other components (resistors, transistors, etc..) are connected.
The esp is powered from USB for now, not the 5V from the P1 port.

I get the following output:

```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/dsmr-node.yaml...
INFO Starting log output from 192.168.31.16 using esphome API
INFO Successfully connected to dsmr-node @ 192.168.31.16 in 0.109s
INFO Successful handshake with dsmr-node @ 192.168.31.16 in 2.585s
[12:27:59][I][app:115]: ESPHome version 2025.5.2 compiled on Jun 11 2025, 12:26:46
[12:27:59][C][wifi:600]: WiFi:
[12:27:59][C][wifi:428]:   Local MAC: 84:CC:A8:B3:75:54
[12:27:59][C][wifi:433]:   SSID: [redacted]
[12:27:59][C][wifi:436]:   IP Address: 192.168.31.16
[12:27:59][C][wifi:439]:   BSSID: [redacted]
[12:27:59][C][wifi:441]:   Hostname: 'dsmr-node'
[12:27:59][C][wifi:443]:   Signal strength: -66 dB ▂▄▆█
[12:27:59][C][wifi:447]:   Channel: 6
[12:27:59][C][wifi:448]:   Subnet: 255.255.255.0
[12:27:59][C][wifi:449]:   Gateway: 192.168.31.1
[12:27:59][C][wifi:450]:   DNS1: 62.179.104.196
[12:27:59][C][wifi:451]:   DNS2: 213.46.228.196
[12:27:59][C][logger:224]: Logger:
[12:27:59][C][logger:225]:   Max Level: DEBUG
[12:27:59][C][logger:226]:   Initial Level: DEBUG
[12:27:59][C][logger:228]:   Log Baud Rate: 0
[12:27:59][C][logger:229]:   Hardware UART: UART0
[12:27:59][C][uart.arduino_esp8266:118]: UART Bus:
[12:27:59][C][uart.arduino_esp8266:120]:   RX Pin: GPIO13
[12:27:59][C][uart.arduino_esp8266:122]:   RX Buffer Size: 3072
[12:27:59][C][uart.arduino_esp8266:124]:   Baud Rate: 115200 baud
[12:27:59][C][uart.arduino_esp8266:125]:   Data Bits: 8
[12:27:59][C][uart.arduino_esp8266:126]:   Parity: NONE
[12:27:59][C][uart.arduino_esp8266:127]:   Stop bits: 1
[12:27:59][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[12:27:59][C][switch.gpio:068]: GPIO Switch 'status_led'
[12:27:59][C][switch.gpio:090]:   Restore Mode: always OFF
[12:27:59][C][switch.gpio:031]:   Pin: GPIO2
[12:27:59][C][P1Mini:489]: P1 Mini component
[12:27:59][W][P1Mini:146]: Unknown data format (0xe7). Resetting.
[12:27:59][D][switch:016]: 'status_led' Turning OFF.
[12:27:59][D][switch:055]: 'status_led': Sending state OFF
[12:27:59][W][P1Mini:482]: Discarding: 9fe7fe3ec7c7c7e7ffe7e7ffffe7c7dfffcffcfceefff7c7f7f7f7e7e7ffc7ff
[12:27:59][W][P1Mini:482]: Discarding: ffcff8c7f7dff7c7f7c7e7f7ffc7f7dfdfdff7f7ffffe7ffeffcfcf8c7f7fff7
[12:27:59][W][P1Mini:482]: Discarding: f7dff7dfe7e7ffffc7ffc7ffffffdfffc7ffc7ffffffe7ffe7ffffffffffe7ff
[12:27:59][W][P1Mini:482]: Discarding: c7ffdffff7ffdfffffcfeec7f7dff7e7f7dfe7c7c7dfdffffff7ffdfe7f7fffc
[12:27:59][W][P1Mini:482]: Discarding: effcfceec7f7dff7e7f7f7e7c7c7dfe7dff7f7e7e7f7f7fffceffcfcfec7f7f7
[12:27:59][W][P1Mini:482]: Discarding: f7e7f7dfe7c7c7c7c7c7c7f7c7c7c7f7fffcfffcfceec7f7f7f7e7f7f7e7c7c7
[12:27:59][W][P1Mini:482]: Discarding: c7c7c7c7f7c7c7c7f7fffceffcfcf8c7effff7f7dfe7f7c7e7c7c7c7f7ffffcf
[12:27:59][W][P1Mini:482]: Discarding: eec7f7dff7fff7c7e7c7c7f7c7e7fff7ffeffcfceec7f7f7f7fff7c7e7c7c7f7
[12:27:59][W][P1Mini:482]: Discarding: c7c7c7f7ffeffcfcf8c7f7fff7f7fff7f7dfe7c7c7c7dfffffffcff8c7f7fff7
[12:27:59][C][captive_portal:089]: Captive Portal:
[12:27:59][C][web_server:285]: Web Server:
[12:27:59][C][web_server:286]:   Address: dsmr-node.local:80
[12:27:59][W][P1Mini:482]: Discarding: f7fff7ffe7c7c7c7c7ffffffcfeec7f7fffff7fffff7c7e7e7ffe7c7ffc7f7ff
[12:27:59][C][mdns:120]: mDNS:
[12:27:59][C][mdns:121]:   Hostname: dsmr-node
[12:27:59][W][P1Mini:482]: Discarding: f7f7fff7dffffffffcfbeeee77ffc7ffffffc7e7ffefc7c7c7c7c7fffff7dff7
[12:27:59][C][esphome.ota:073]: Over-The-Air updates:
[12:27:59][C][esphome.ota:074]:   Address: dsmr-node.local:8266
[12:27:59][C][esphome.ota:075]:   Version: 2
[12:27:59][C][esphome.ota:078]:   Password configured
[12:27:59][C][safe_mode:018]: Safe Mode:
[12:27:59][C][safe_mode:019]:   Boot considered successful after 60 seconds
[... 85 lines cut ...]
[12:28:02][W][P1Mini:482]: Discarding: f7e7f7dfe7c7c7c7c7c7c7f7c7c7c7f7fffceffcfceec7f7f7f7e7f7f7e7c7c7
[12:28:02][W][P1Mini:482]: Discarding: c7c7c7c7f7c7c7c7f7fffceffcfcf8c7effff7f7dfe7f7c7e7c7c7c7f7ffffcf
[12:28:02][W][P1Mini:482]: Discarding: eec7f7dff7fff7c7e7c7c7f7c7e7f7f7ffeffcfceec7f7f7f7fff7c7e7c7c7f7
[12:28:02][W][P1Mini:482]: Discarding: c7c7c7f7ffeffcfcf8c7f7fff7f7fff7f7dfe7c7c7c7dfffffffcff8c7f7fff7
[12:28:02][W][P1Mini:482]: Discarding: f7fff7ffe7c7c7c7c7ffffffcffec7f7fffff7fffff7c7e7e7ffe7c7ffc7f7ff
[12:28:02][W][P1Mini:482]: Discarding: f7f7fff7dffffffffcfbeeee77ffc7ffffffc7e7ffefc7c7c7c7c7fffff7dff7
[12:28:02][W][P1Mini:482]: Discarding: f7ffeffcfb77ffdfdfdfffffe7f7ffffefc7c7c7c7c7c7c7dfe7dff7ffeffcfb
[12:28:02][W][P1Mini:482]: Discarding: 77ffdfdfdfe7f7f7dfffffefc7c7c7c7c7c7dfe7f7fff7ffeffcf8f8f8eedfc7
[12:28:02][W][P1Mini:482]: Discarding: c7c7c7c7c7ffefc7c7c7c7c7c7c7ffdffff7ffeffcfceec7f7fff7f7fff7f7c7
[12:28:02][W][P1Mini:482]: Discarding: e7c7c7c7dfdfffffcfeec7f7fff7f7fff7f7c7e7c7c7c7c7ffffffcff8c7f7ff
[12:28:02][W][P1Mini:482]: Discarding: f7f7dffff7c7e7ffffcfeec7f7fff7f7fff7c7e7f7fff7f7e7f7f7effcfceec7
[12:28:02][W][P1Mini:482]: Discarding: f7ffdff7fff7c7e7c7c7dff71feffcfceec7f7f7dff7fff7c7e7c7c7f7c7e7f7
[12:28:02][W][P1Mini:482]: Discarding: f7ffeffcfceec7f7f7f7f7fff7c7e7c7c7f7c7c7c7f7ffeffcfcf8dff7f7e7f7
[12:28:02][W][P1Mini:482]: Discarding: dff7c7e7c7c7ffffffcff8dff7fff7f7dff7c7e7e7ffffc7ffc7fff7ffe7ffc7
[12:28:02][W][P1Mini:482]: Discarding: ffc7fff7ffc7ffc7fffffff7ffdfffc7ffc7fff7ffc7ffffcff8dff7f7e7f7f7
[12:28:02][W][P1Mini:482]: Discarding: f7dfe7f7ffc7f7dfdfdff7f7ffc7ffffefc7fff7f7e7f7e7f7f7f7fff7effcfc
[12:28:02][W][P1Mini:482]: Discarding: ecef1fe6fcfc
[12:28:02][D][switch:012]: 'status_led' Turning ON.
[12:28:02][D][switch:055]: 'status_led': Sending state ON
[12:28:03][W][P1Mini:146]: Unknown data format (0xe7). Resetting.
[12:28:03][D][switch:016]: 'status_led' Turning OFF.
[12:28:03][D][switch:055]: 'status_led': Sending state OFF
```

Any idea what could be going wrong here?

thedevleon, 2025-06-11:
Relevant sections from the config:
```
substitutions:
  device_name: dsmr-node

esphome:
  name: ${device_name}
  friendly_name: ${device_name}

esp8266:
  board: d1_mini

# Disable logging on UART so we can use the hardware UART
logger:
  baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: [redacted]

ota:
  platform: esphome
  password: [redacted]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "dsmr-node Fallback Hotspot"
    password: [redacted]

captive_portal:

web_server:
  port: 80

switch:
  - platform: gpio
    id: status_led
    pin:
      number: D4
      inverted: true

uart:
  id: my_uart_1
  rx_pin:
    number: D7
    inverted: true
    mode: INPUT_PULLUP
  baud_rate: 115200
  rx_buffer_size: 3072

external_components:
  - source: github://Beaky2000/esphome-p1mini@main
  

p1_mini:
  - id: p1_mini_1
    uart_id: my_uart_1
    minimum_period: 0s       # Should be 0 (zero) if the RTS signal is not used.
    buffer_size: 3072        # Needs to be large enough to hold one entire update from the meter.
    secondary_p1: false
    on_ready_to_receive:
      then:
        # - switch.turn_on: p1_rts
        - switch.turn_on: status_led
[... 224 lines cut ...]
    name: "Current Phase 1"
    icon: "mdi:lightning-bolt"
    unit_of_measurement: A
    accuracy_decimals: 1
    device_class: "current"
    state_class: "measurement"
  - platform: p1_mini
    p1_mini_id: p1_mini_1
    obis_code: "51.7.0"
    name: "Current Phase 2"
    icon: "mdi:lightning-bolt"
    unit_of_measurement: A
    accuracy_decimals: 1
    device_class: "current"
    state_class: "measurement"
  - platform: p1_mini
    p1_mini_id: p1_mini_1
    obis_code: "71.7.0"
    name: "Current Phase 3"
    icon: "mdi:lightning-bolt"
    unit_of_measurement: A
    accuracy_decimals: 1
    device_class: "current"
    state_class: "measurement"
```

Beaky2000, 2025-06-11:
Looks like a case of [insufficient pull-up resistor](https://github.com/Beaky2000/esphome-p1mini/blob/main/docs/troubleshooting.md#insufficient-pull-up-resistor)

thedevleon, 2025-06-11:
I got it working thanks to the comment [here](https://github.com/Beaky2000/esphome-p1mini/issues/33#issuecomment-2178721247) - now I got pretty much all the values I needed.

Is there any way to parse the gas as well?
```
[19:18:38][D][P1Mini:282]: No sensor matched line '0-1:24.2.1(250611191503S)(05264.591*m3)'
```